Edzelf/ESP32-Radio #76: volume set does not work from web interface

AST-esp-rasp, 2018-01-07:
hello I can't set the volume from the web interface.
D: Command client available
D: handlehttp started
D: Get command is: upvolume=2
D: Send reply for upvolume=2
D: Command: upvolume with parameter 2
this is what i see in the serial interface but nothing happens to the headphone.
Mute does not work as well.
D: Command client available
D: handlehttp started
D: Get command is: mute
D: Send reply for mute
D: Command: mute with parameter 0
this is what i see in the serial interface but nothing happens to the headphone.

Edzelf, 2018-01-08:
Looks like there is no connection to the VS1053.
Is the radio playing?

AST-esp-rasp, 2018-01-11:
the radio is playing but the sound is awful and then it interrupts

Edzelf, 2018-01-11:
Please show the full debug logging at start-up.