Edzelf/ESP32-Radio #77: ISP Slow start TCP changing channels - only affects some ISPs

birdwes, 2018-01-14:
Hi, 

I would first like to say thank you for this brilliant project.  I assembled just an ESP-32S and the square blue VS1053 board this afternoon as a rough prototype with wire-wrap and it worked first time.  I have however taken it out to show friends for further testing.  

It seems that with my home ISP that there is some slow start TCP algorithm that makes it keep jumping channels when you try to select a different preset.  At other places I tried it, this was not a problem, they tended to select first time.  With my ISP you have to try to select a preset about six or seven times within a minute, otherwise it just keeps jumping from preset to preset several times again.  Once it has locked on to a preset it is fine.

I have not had a chance to look at the code yet, but it seems too eager to try a new preset.

Thanks,

birdwes, 2018-01-14:
P.S. it's not that I have a slow (for here) connection, I get 22Mb/s download on a line test, which is over twice the speed I tried at the other places, which work better.

Edzelf, 2018-01-14:
Try to increase the time-out in line 1603.  Change:
  if ( ( morethanonce > 0 ) ||                // Happened more than once?
into:
  if ( ( morethanonce > 1 ) ||                // Happened more than twice?

birdwes, 2018-01-15:
Thanks Ed for responding so quickly.  

I will try this soon.

Is there a way to PM on this place.  I need to to send you a message about other new ideas and share new code.

birdwes, 2018-01-15:
to be specific, a JSON parser with no malloc, that can stream and select from objects larger than the free RAM.

Edzelf, 2018-01-15:
You can find my email address at https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino (line 3).