Edzelf/ESP32-Radio #78: Connecting to open WiFi with additional credentials

vadpol, 2018-01-15:
Hello,

I have a question regarding some open public wifi networks which still require user ID and password while connecting to them (i.e. hotels, restaurants, etc.). Is there any possibility to get connected to such a network in case these credentials are known in before (uploaded via sketch) or provided on spot (via webinterface)? Many thanks!

Edzelf, 2018-01-15:
Yes.  Read the documentation.  You can provide the credentials through the webinterface.

vadpol, 2018-01-15:
Actually this was my question: at some places when you connect to 'open' WiFi network you see the webpage that asks for user ID and password or to make some ticks on the page to get connected to Internet.  An example: 'WiFi in the train' in the NL. So it is not possible to connect at all on physical level in such a case or is it a matter of the future software development?

Edzelf, 2018-01-15:
It is not my plan to develop this.  It might be very hard to hack the various types of open networks.

vadpol, 2018-01-17:
Hi Ed,

Ok, that's your right. Thanks anyway! I'm just testing the new features of ESP32 based web radio, it seems they are more or less working as prescribed. Can I ask you which preferences I have to fill in in init file in order to control mp3 player directly via IR? I think the just moving through the mp3 file list up and down and play/stop commands would be enough. I have all the codes from the buttons on my remote and partly use them already to switch in between stations and volume control, but direct control of the mp3 player via IR would be also a great feature as to me. Thanks a lot!

Edzelf, 2018-01-17:
In the current version only "ir_xxxx = mp3track=0" is supported.  This command will switch to MP3-player and select the (next) random track.

vadpol, 2018-01-17:
Many thanks Ed! This is something to test now at least. Perhaps you will extend it in the future, this would be great indeed! 

Here is my setup, tried to make it as compact as possible, used bare ESP32 module on adapter board, LiPo battery and charger board plus power 2x3W amplifier and IR sensor are under display and VS1063 boards.

[image]

Edzelf, 2018-01-17:
Looks great!

vadpol, 2018-01-17:
Here it is in more detail, the 4 pin header for programming is next to ESP32 module in the bottom, the header at the upper right side of the board is for connection of the encoder, L and R audio in from VS1053, connected Power/Charge USB/BAT switch (had to use an additional 5V relay for that purpose), L and R speaker out from amplifier. The LiPo battery is two banks of 1000 mAh 3,7 volt connected in parallel, not separated for ESP/VS boards and amplifier, but work perfectly well, no sound distortion noticed so far.
[image]

closed 2018-01-19 by vadpol