Edzelf/ESP32-Radio #79: Looks like an initialization bug

Mammux, 2018-01-20:
I think this must be a bug. Pin numbers can be set to -1 or whatever, if they are not found in NVS.

https://github.com/Edzelf/ESP32-Radio/blob/0f5cdda7890f9543adb77d8dfdfb94eb8aebdec4/Esp32_radio.ino#L2268

Edzelf, 2018-01-21:
I do not understand.  What is the problem?

DK5RM, 2018-01-21:
It seems I have a similar problem. Sketch compiles with no errors, but the ESP-32 Radio  does not start. No WiFI, TFT or VS1053. I think the initialization failed.
All the necessary pins are set to -1 (see debug listing below)
Is there something I overlooked? I am not very experienced in ESP32 programming.
Any advice is highly appreciated.

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 02 Jan 2018 08:50:00 GMT.  Free memory 171952
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir pin set to -1
D: pin_enc_clk pin set to -1
D: pin_enc_dt pin set to -1
D: pin_enc_sw pin set to -1
D: pin_tft_cs pin set to -1
D: pin_tft_dc pin set to -1
D: pin_sd_cs pin set to -1
D: pin_vs_cs pin set to -1
D: pin_vs_dcs pin set to -1
D: pin_vs_dreq pin set to -1
D: pin_shutdown pin set to -1
D: pin_spi_sck pin set to -1
D: pin_spi_miso pin set to -1
D: pin_spi_mosi pin set to -1
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
........
........
........

Mammux, 2018-01-21:
I had the same problem as DK5RM.

Edzelf, 2018-01-21:
Go to the config page in thew web interface.  Check if you have lines like this:
pin_enc_clk = 25     # GPIO Pin number for rotary encoder CLK
pin_enc_dt = 26      # GPIO Pin number for rotary encoder DT
pin_enc_sw = 27      # GPIO Pin number for rotary encoder SW
pin_ir = 35          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21       # GPIO Pin number for SD card CS
pin_tft_cs = 15      # GPIO Pin number for TFT CS
pin_tft_dc = 2       # GPIO Pin number for TFT DC
pin_vs_cs = 5        # GPIO Pin number for VS1053 CS
pin_vs_dcs = 16      # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 4      # GPIO Pin number for VS1053 DREQ
#
Note that the keywords differ from earlier versions.

Edzelf, 2018-01-21:
Is there a line like:
pin-vs-dreq=5
in your configuration?
There should be one!

DK5RM, 2018-01-22:
Hello Ed,
thank's for looking into this. Hope I can help.
Below is a copy of the definitions in the defaultprefs.h from Github, which I didn't change.
This is consistent with the wireing instructions in the ESP32_radio.ino file and with my hardware setup.

pin_vs_cs = 5                                          # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"

I think the definitions are ok, but nevertheless  initialization fails.  Pins are set to -1, so the program will not start properly.

Edzelf, 2018-01-22:
The settings in defaultprefs.h do NOT have any effect on the preferences used by the radio.
You may however transfer the settings in defaultprefs to the actual settings by pressing the "Default" button in the config screen.  Then edit these setting (do not forget the WiFi credentials)  and presss the "Save" button.

There is a nice testprogram at [https://github.com/Edzelf/ESP32-Show_nvs_keys](url) that show all your preferences in the ESP32.  These are the ones that are actually used by the ESP32-Radio.

DK5RM, 2018-01-22:
Hi Ed,
I tried your tool and it shows, that NVS settings are not correct (listing below).
However, one can not edit via the Web interface, because WiFi DOES NOT START due to missing nvs settings. Same problem as Mammux posted yesterday.
Now I struggle to write the necessary  pin assingnments to nvs or at least start the web interface.

D: Partition nvs found, 20480 bytes
D: Namespace ID of ESP32Radio is 4
D: Key 000: gpio_00
D: Key 002: gpio_12
D: Key 004: gpio_13
D: Key 006: mqttbroker
D: Key 008: mqttport
D: Key 010: mqttuser
D: Key 012: mqttpasswd
D: Key 014: mqtttpreset
D: Key 016: wifi_00
D: Key 018: wifi_01
D: Key 020: volume
D: Key 022: toneha
D: Key 024: tonehf
D: Key 026: tonela
D: Key 028: tonelf
D: Key 030: preset
D: Key 032: preset_00
D: Key 036: preset_01
D: Key 040: preset_02
D: Key 044: preset_03
D: Key 048: preset_04
D: Key 052: preset_05
D: Key 056: preset_06
D: Key 059: preset_07
D: Key 063: preset_08
D: Key 066: preset_09
D: Key 070: preset_10
D: Key 074: preset_11
D: Key 078: clk_server
D: Key 080: clk_offset
D: Key 082: clk_dst

Edzelf, 2018-01-22:
You may initialize the NVS wih /tools/Esp32_radio_init.ino.  In the document:
```
Alternative way of configuration.
An extra sketch "Esp32_radio_init" is supplied as an alternative to initialize the preferences
(in Non-Voltile Storage of the ESP32).  Just change lines 39 and 40 (the specs for WiFi networks)
to match your network(s).
Upload and run the sketch once and then load the ESP32-Radio.
```

DK5RM, 2018-01-22:
It is working!
After wiriting the pin preferences with "Esp32_radio_init"  to the NVS and reloading ESP_radio it started WiFi and I was able to set up the radio properly. Great radio!
Thank's Ed for your continous support and sharing this program with us.

Edzelf, 2018-01-22:
Great!  I will use default pinnumbers (as described in the doc) in the next release.