Edzelf/ESP32-Radio #80: On-screen battery remained life indicator

vadpol, 2018-01-21:
Hi Ed, 
just came across to an idea to implement the above indicator since there are few free analogue pins sill available at ESP32 module. As soon as battery voltage is changing from 4,2V (fully charged) to say 3,6V it might be possible to make a resistor voltage divider at ESP32's assigned analogue pin. The visualization might be in the form of percent or bar like sound bar. Would it be possible to work out this piece of code? Many thanks!

Edzelf, 2018-01-22:
I implemented this as a optional feature of the ESP32-Radio.  See documentation.

vadpol, 2018-02-01:
Hi Ed,

Many thanks for your excellent job and fast reaction! Will try it out, it may really be useful!

closed 2018-02-01 by vadpol