Edzelf/ESP32-Radio #81: How to debug lack of sound output

Mammux, 2018-01-22:
I'm fairly certain I've wired everything up properly, and according to serial and web output it seems to connect to the appropriate radio stations etc. But there is no sound from my VS1053. I use the model from LCSOFT, it's pictured in the documentation. The line out is connected to my stereo, but I also looked at it on a scope. I'm uncertain how to debug this further - maybe I'll have to disconnect the VS1053 and test it from an arduino? Log from the application (I entered 'status' and 'test' a couple of times after startup):

```
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Tue, 02 Jan 2018 08:50:00 GMT.  Free memory 171592
D: Partition nvs found, 20480 bytes
D: Read 29 keys from NVS
D: pin_ir pin set to 35
D: pin_enc_clk pin set to 25
D: pin_enc_dt pin set to 26
D: pin_enc_sw pin set to 27
D: pin_tft_cs pin set to 15
D: pin_tft_dc pin set to 2
D: pin_sd_cs pin set to 21
D: pin_vs_cs pin set to 5
D: pin_vs_dcs pin set to 16
D: pin_vs_dreq pin set to 4
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: Start TFT
D: SD Card Mount Failed!
D: Added NorSko40 to list of networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 3
D:  1 - NorSko40                  Signal: -52 dBm, Encryption WPA_PSK, Acceptable
D:  2 - private                   Signal: -52 dBm, Encryption  WEP, 
D:  3 - Cassiopeia                Signal: -82 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 7
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 100
D: Slow SPI, Testing VS1053 read/write registers...
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Try WiFi NorSko40 / **********
D: Connected to NorSko40
D: IP = 192.168.2.5
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 16:01:32
D: STOP requested
D: New preset/file requested (7/0) from 205.164.62.15:10032
D: Connect to new host 205.164.62.15:10032
D: Connect to 205.164.62.15 on port 10032, extension /
D: Connected to server
D: Switch to HEADER
D: Headerline: icy-br: 32
D: Headerline: icy-pub: 1
D: Headerline: icy-description: 1.FM - Radio Gaia
D: Headerline: icy-audio-info: bitrate=32;samplerate=44100;channels=2
D: Headerline: icy-url: http://1.fm
D: Headerline: Instance-id: 3632553544703fea35d133f849e6e346
D: Headerline: Cache-Control: no-cache
D: Headerline: Server: AIS Streaming Server 7.7.2
D: Headerline: icy-genre: Chill
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint: 8192
D: Headerline: Pragma: no-cache
D: Headerline: icy-name: 1.FM - Radio Gaia
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.
D: Switch to DATA, bitrate is 32, metaint is 8192
D: Metadata block 64 bytes
D: Streamtitle found, 49 bytes
D: StreamTitle='DAVID ARKENSTONE - A SPECIAL PLACE';
D: Command: status with parameter 0
D: 1.FM - Radio Gaia - DAVID ARKENSTONE - A SPECIAL PLACE
D: Command: test with parameter 0
D: Stack CPU0 is 996
D: Stack CPU1 is 6488
D: Free memory is 100000, chunks in queue 0, stream 0, bitrate 50 kbps
D: Command: test with parameter 0
D: Stack CPU0 is 996
D: Stack CPU1 is 6488
D: Free memory is 100000, chunks in queue 0, stream 0, bitrate 32 kbps
D: Command: test with parameter 0
D: Stack CPU0 is 996
D: Stack CPU1 is 6488
D: Free memory is 100000, chunks in queue 0, stream 0, bitrate 32 kbps
```

Log from show_nvs_keys:

```
D: Partition nvs found, 20480 bytes
D: Namespace ID of ESP32Radio is 1
D: Key 002: pin_vs_dcs
D: Key 005: pin_vs_dreq
D: Key 044: preset_00
D: Key 048: preset_01
D: Key 052: preset_02
D: Key 056: preset_03
D: Key 060: preset_04
D: Key 064: preset_05
D: Key 068: preset_06
D: Key 071: preset_07
D: Key 075: preset_08
D: Key 078: preset_09
D: Key 082: preset_10
D: Key 086: preset_11
D: Key 090: toneha
D: Key 092: tonehf
D: Key 094: tonela
D: Key 096: tonelf
D: Key 100: wifi_00
D: Key 102: pin_enc_clk
D: Key 105: pin_enc_dt
D: Key 108: pin_enc_sw
D: Key 111: pin_ir
D: Key 114: pin_sd_cs
D: Key 117: pin_tft_cs
D: Key 119: pin_tft_dc
D: Key 121: pin_vs_cs
D: Key 043: preset
D: Key 045: volume
```

Edzelf, 2018-01-22:
This is an aacp station.  Are you sure you have a VS1053B chip in your mp3-module?
Try an mp3 station first, like us2.internet-radio.com:8050

DK5RM, 2018-01-23:
Hi Mammux,
I am using the same VS1053 Module as you. Mine also had no audio output. Reason was a bad / intermittent  connection inside the 3.5mm audio jack.  Fixed it by directly soldering wires to the traces.

Edzelf, 2018-01-23:
I think this is not the problem.  In the logging you see that there are no chunks in the queue.  The computed bitrate is correct (32 kbps).  So the VS1053 eats up all the incoming data without processing.  My conclusion is that the stream is not recognized by the chip, therefore it migth be a VS1003, which is not capable of decoding AACP.

Mammux, 2018-01-23:
According to the print on the chip it's a VS8053B. I tried us2.internet-radio.com:8050 as well, with the same result. There is a slight "clicking" if I turn up the volume, but no real sound. The faint clicking goes away if I stop the player in the web interface.

Edzelf, 2018-01-23:
A VS8053B is not an VS1053B.  That may cause the problem.

Mammux, 2018-01-23:
Ah!

"VS8053b firmware uses laser fuse bits to disable MP3, WMA, AAC and Midi decoders. Otherwise
the functionality and performance for VS1053b and VS8053b is exactly the same."

Mammux, 2018-01-23:
Are there any webradio stations that stream Ogg Vorbis / FLAC / WAV? :)

Edzelf, 2018-01-24:
Try www.music.helsinki.fi/tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav.

Mammux, 2018-01-24:
That didn't work either. I'll try another VS1053b module, that seems to have the correct chip. I'll have to do some soldering to connect it though.

markome, 2019-12-02:
Hi there!
First, I would like to thank you for the great project!
However, I am also encountering "no sound" problem.
I am using v1053b board.
I am pretty sure, I have wired everything correctly since I am able to hear the sound when i am using the code of the project 
http://educ8s.tv/esp32-internet-radio (updated with correct GPIO pins)
But using the Edzelf/ESP32-Radio, I am not able to produce a sound. 

I am attaching the debug output to the console:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 293956
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 48 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added LDOS to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
[... 104 lines cut ...]
D: Duration mp3loop 85
D: Command client available
D: handlehttp started
D: Get command is: station=www.music.helsinki.fi/tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav
D: Send reply for station=www.music.helsinki.fi/tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav
D: Command: station with parameter www.music.helsinki.fi/tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav
D: STOP requested
D: Stopping client
D: Connect to new host www.music.helsinki.fi/tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav
D: Connect to www.music.helsinki.fi on port 80, extension /tmt/opetus/uusmedia/esim/a2002011001-e02-ulaw.wav
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Switch to HEADER
D: Headerline: Server: Apache/2.4.6 (Red Hat Enterprise Linux) OpenSSL/1.0.2k-fips PHP/5.4.16 mod_perl/2.0.10 Perl/v5.16.3
D: Headerline: Last-Modified: Fri, 01 Feb 2002 07:23:18 GMT
D: Headerline: Accept-Ranges: bytes
D: Headerline: Content-Length: 4790318
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/x-wav
D: audio/x-wav seen.
D: Switch to DATA, bitrate is 0, metaint is 0
D: Command client available
D: handlehttp started
D: HTTP input shorter than expected

Edzelf, 2019-12-02:
I see in the logging:
D: Command: volume with parameter 0
So add a line containing:
`volume=76`
in the configuration (webinterface)

markome, 2019-12-02:
Thank you!
It seems, that I have a problem with the rotary encoder.

My setup is constantly decreasing volume, hence no audio output after I put on the headphones.
I have commented out //dbgprint ( "Rotation count %d", rotationcount ) ; in the code, since I was getting a lot of "D: Rotation count -1" lines in the console window.

Edzelf, 2019-12-02:
Use the winamp trick described in the document.