Edzelf/ESP32-Radio #82: Instaling ESP32-Radio

amigafan, 2018-02-01:
I'm having trouble installing. Should both Esp32_radio_init.ino and Esp32_radio.ino be downloaded to ESP32? And in what order? It's not yet successful. Any help received with thanks.

locki-cz, 2018-02-01:
Hi, i have same question :) I started with version  28-08-2017 and all worked fine with just flashing Esp32_radio. Now I tshooted why it not working with new version and i find out i must first flash Esp32_radio_init and then Esp32_radio. Another problem is configuration, it not working properly, it save random only one line :( then settings must be edited in init and flash again and flash radio again ... maybe I am doing something wrong ... another problem is I edited few lines for ST7735 display (in pull request) and it worked on old version .... now on new one just white screen, wiring didn't changed ... 

I hope helped you a little bit. Try and let know if it is same for you.

Edzelf, 2018-02-01:
@amigafan: The main function of Esp32_radio_init.ino is to get the network started and to create the NVS preferences.  So the WiFi settings are important here.  Just flash and run the sketch once for a new ESP32.
After that, you can flash the Esp32_radio.ino script and use the web interface to alter the configuration.

locki-cz, 2018-02-01:
But confuguration via web interface not working, properly, it not case whole config, when i just edit SSID and remove radio stations i using only 4 stations ... it si shorter than "demo" how to tshoot it?

Edzelf, 2018-02-01:
@locki-cz: Saving the configuration may fail if the contents are too big for the NVS area.  The NVS area is about 20 kB.  Lines are limited to 150 characters.
Show me your config, so I can try to reproduce it.

perlix, 2018-02-01:
I'm using version 22 Jan 2018 (on a LOLIN32 Pro) and can also not save configurations via web, even if the content is smaller than the original. Saving erases my wifi credentials. 
Configuration via Esp32_radio_init.ino works great though, so I don't really mind. I love this radio!

locki-cz, 2018-02-01:
@Edzelf nothing special at all, only 3 buttons 4 stations and 2 ssid, it have less lines then default Esp32_radio_init.ino i will provide it latter from home :) And yes it is awesome project! :)

amigafan, 2018-02-01:
[image]
So, I manage to install ESP32-Radio, but the web interface react very sluggishly, unlike ESP12 radio. How can it be? It's a faster processor! Thanks for all the help.

Edzelf, 2018-02-01:
Strange.  The web interface should be much faster than the ESP8266 version.

locki-cz, 2018-02-01:
my config i just change ssid and passwords :)

`//********************************************************************************************
// Initialize the preferences op ESP32-radio.  They can be edited later by the webinterface. *
//********************************************************************************************
// 27-04-2017, ES: First set-up, derived from preferences example sketch.                    *
//********************************************************************************************

#include <Preferences.h>

// Note: Namespace name is limited to 15 chars.  Same name must be used in ESP32-radio.
#define NAME "ESP32Radio"
// Adjust size of buffer to the longest expected string for nvsgetstr
#define NVSBUFSIZE 150

Preferences preferences ;         // Instance of Preferences

//******************************************************************************************
//                              S E T U P                                                  *
//******************************************************************************************
void setup()
{
  String       str ;
  
  Serial.begin ( 115200 ) ;
  Serial.println() ;
  preferences.begin ( NAME, false ) ;                 // Open for read/write
  preferences.clear() ;                               // Remove all preferences under opened namespace
  // Store the preferences for ESP32-radio
  preferences.putString ( "gpio_13",  "uppreset = 1" ) ;
  preferences.putString ( "gpio_12",  "upvolume = 2" ) ;
  preferences.putString ( "gpio_14",  "downvolume = 2" ) ;

  preferences.putString ( "mqttbroker",   "none" ) |
  preferences.putString ( "mqttport",     "1883" ) |
  preferences.putString ( "mqttuser",     "none" ) |
  preferences.putString ( "mqttpasswd",   "none" ) |
  preferences.putString ( "mqtttpreset",  "none" ) |
  //
  preferences.putString ( "wifi_00", "Home/*********" ) |
  preferences.putString ( "wifi_01", "MY Guest/*********" ) |

//  # Enter your WiFi network specs here:
  //
  preferences.putString ( "volume", "100" ) |
  preferences.putString ( "toneha", "0" ) |
  preferences.putString ( "tonehf", "0" ) |
  preferences.putString ( "tonela", "0" ) |
  preferences.putString ( "tonelf", "0" ) |
  //
  preferences.putString ( "preset", "0" ) |
  preferences.putString ( "preset_00", "ice.abradio.cz:8000/helax128.mp3.m3u      #  0 - Helax" ) ;
  preferences.putString ( "preset_01", "icecast3.play.cz:80/evropa2-128.mp3       #  1 - Evropa2" ) ;
  preferences.putString ( "preset_02", "icecast5.play.cz:8000/impuls128.mp3       #  2 - Impuls" ) ;
  preferences.putString ( "preset_03", "mp3stream7.apasf.apa.at:80                #  3 - OE3" ) ;
  //
  preferences.putString ( "clk_server", "pool.ntp.org" ) ;        // Setting for Time Of Day clock on TFT
  preferences.putString ( "clk_offset", "1" ) ;
  preferences.putString ( "clk_dst",    "1" ) ;
  preferences.end() ;
  delay ( 1000 ) ;
}

//******************************************************************************************
//                              L O O P                                                    *
//******************************************************************************************
void loop()
{
  Serial.println ( "ESP32_radio_init completed..." ) ;
  delay ( 10000 ) ;
}

`

amigafan, 2018-02-01:
I just tried to use my mobile phone as a Wi-Fi hotspot and then connect the ESP32-radio as well as my tablet. Then there is no delay when I maneuver around the web page. So there must be something in my LAN network that gives a delay. By the way, the upper left button "ESP Radio" apparently has no function!

locki-cz, 2018-02-01:
problem with LCD solved, missing in initial sketch:

preferences.putString ( "pin_tft_cs",  "15" ) ;
preferences.putString ( "pin_tft_dc",  "2" ) ;

from debug was visible:
defined as -1 (not defined)

:)