Edzelf/ESP32-Radio #86: Spectrum analyzer

MirkoDalmonte, 2018-02-12:
Spectrum analyzer is not essential but it is beautiful :)

https://m.youtube.com/watch?v=mSrDDmvLRfI

locki-cz, 2018-02-14:
Very nice!!! How you added it? How is possible edit custom display positions and text ? Can you kick me little bit?

I want to my project add when you connect it to power it boot, connect wifi etc and stay stopped and just show date, time, day, after button press it start play radio...

MirkoDalmonte, 2018-02-14:
Hi,
add spectrum analyzer is a bit complex..
I have loaded in the vs1053 the last patch and the dedicated plugin.
Every 100ms I read the values ​​directly in vs1053 and display them. This process is quite fast so the radio works well even at 320.
Customize the display is very simple, there is a struct (tftdata) to modify existing sectors or add others... so in the displayinfo() function you can customize each sector by comparing the index of this struct.
Calling tftset (index, char*) or tftset (index, string) you can display in a specific sector.
Spectrum analyzer not use these functions, it uses the display directly, so its space is reserved in the struct (90-128).
To leave stopped the radio at the start you must set currentpreset = ini_block.newpreset in setup ().

koskee, 2018-04-22:
Any chance you'd post your code for loading the patch/plugin? I've been toying with the idea of doing this for a while, but haven't had a chance to dive in. 
Also, I have a question about how you display the output... When writing the bars to the screen, how did you manage to avoid overwriting the text the display? Does it automatically keep track of the text or do you have it buffered somewhere or does the screen differentiate between text/fill commands.. ?
To me it seems like the text would be covered up unless you replace it every frame, which would be a lot of updating...

MirkoDalmonte, 2018-04-23:
Hi koskee,
yes.... the code and the answer to your questions as soon as I have time.... soon...

koskee, 2018-04-23:
Awesome. I'll keep my eyes open for an update in that case. I'm also not in any rush, but thanks in advance.

koskee, 2019-01-02:
@MirkoDalmonte  Did you ever get around to posting code for this? Still very much interested..

MirkoDalmonte, 2019-01-03:
Hi koskee, sorry for the delay...

`//***************************************************************************************************
//*  ESP32_Radio -- Webradio receiver for ESP32, 1.8 color display and VS1053 MP3 module.           *
//*                 By Ed Smallenburg.                                                              *
//***************************************************************************************************
// ESP32 libraries used:
//  - WiFiMulti
//  - nvs
//  - TFT_ILI9163C Sumotoy Version 0.9
//  - ArduinoOTA
//  - PubSubClient
//  - SD
//  - FS
// A library for the VS1053 (for ESP32) is not available (or not easy to find).  Therefore
// a class for this module is derived from the maniacbug library and integrated in this sketch.
//
// See http://www.internet-radio.com for suitable stations.  Add the stations of your choice
// to the preferences in either Esp32_radio_init.ino sketch or through the webinterface.
//
// Brief description of the program:
// First a suitable WiFi network is found and a connection is made.
// Then a connection will be made to a shoutcast server.  The server starts with some
// info in the header in readable ascii, ending with a double CRLF, like:
//  icy-name:Classic Rock Florida - SHE Radio
//  icy-genre:Classic Rock 60s 70s 80s Oldies Miami South Florida
//  icy-url:http://www.ClassicRockFLorida.com
//  content-type:audio/mpeg
//  icy-pub:1
//  icy-metaint:32768          - Metadata after 32768 bytes of MP3-data
//  icy-br:128                 - in kb/sec (for Ogg this is like "icy-br=Quality 2"
//
// After de double CRLF is received, the server starts sending mp3- or Ogg-data.  For mp3, this
// data may contain metadata (non mp3) after every "metaint" mp3 bytes.
// The metadata is empty in most cases, but if any is available the content will be presented on the TFT.
// Pushing an input button causes the player to execute a programmable command.
//
// The display used is a Chinese 1.8 color TFT module 128 x 160 pixels.  The TFT_ILI9163C.h
// file has been changed to reflect this particular module.  TFT_ILI9163C.cpp has been
// changed to use the full screenwidth if rotated to mode "3".  Now there is room for 26
// characters per line and 16 lines.  Software will work without installing the display.
// The SD card interface of the module may be used to play mp3-tracks on the SD card.
//
// For configuration of the WiFi network(s): see the global data section further on.
//
// The VSPI interface is used for VS1053, TFT and SD.
//
// Wiring. Note that this is just an example.  Pins (except 18,19 and 23 of the SPI interface)
// can be configured in the config page of the web interface.
// ESP32dev Signal  Wired to LCD        Wired to VS1053      SDCARD   Wired to the rest
// -------- ------  --------------      -------------------  ------   ---------------
// GPIO16           -                   pin 1 XDCS            -       -
// GPIO5            -                   pin 2 XCS             -        -
// GPIO4            -                   pin 4 DREQ            -       -
// GPIO2            pin 3 D/C           -                     -       -
// GPIO18   SCK     pin 5 CLK           pin 5 SCK             CLK     -
// GPIO19   MISO    -                   pin 7 MISO            MISO    -
// GPIO23   MOSI    pin 4 DIN           pin 6 MOSI            MOSI    -
// GPIO21           -                   -                     CS      -
// GPIO15           pin 2 CS            -                     -       -
// GPI03    RXD0    -                   -                     -       Reserved serial input
// GPIO1    TXD0    -                   -                     -       Reserved serial output
// GPIO34   -       -                   -                     -       Optional pull-up resistor
// GPIO35   -       -                   -                     -       Infrared receiver VS1838B
// GPIO25   -       -                   -                     -       Rotary encoder CLK
// GPIO26   -       -                   -                     -       Rotary encoder DT
// GPIO27   -       -                   -                     -       Rotary encoder SW
// -------  ------  ---------------     -------------------  ------   ----------------
// GND      -       pin 8 GND           pin 8 GND                     Power supply GND
// VCC 5 V  -       pin 7 BL            -                             Power supply
[... 3904 lines cut ...]
      }
      if ( metalinebfx  > ( METASIZ - 10 ) )           // Unlikely metaline length?
      {
        dbgprint ( "Metadata block too long! Skipping all Metadata from now on." ) ;
        metaint = 0 ;                                  // Probably no metadata
      }
      datacount = metaint ;                            // Reset data count
      //bufcnt = 0 ;                                     // Reset buffer count
      datamode = DATA ;                                // Expecting data
    }
  }
  if ( datamode == PLAYLISTINIT )                      // Initialize for receive .m3u file
  {
    // We are going to use metadata to read the lines from the .m3u file
    // Sometimes this will only contain a single line
    metalinebfx = 0 ;                                  // Prepare for new line
    LFcount = 0 ;                                      // For detection end of header
    datamode = PLAYLISTHEADER ;                        // Handle playlist data
    playlistcnt = 1 ;                                  // Reset for compare
    totalcount = 0 ;                                   // Reset totalcount
    dbgprint ( "Read from playlist" ) ;
  }
  if ( datamode == PLAYLISTHEADER )                    // Read header
  {
    if ( ( b > 0x7F ) ||                               // Ignore unprintable cha

MirkoDalmonte, 2019-01-03:
This is the PATCHES file:
`   // PATCHES
    const unsigned short patch[3201] PROGMEM = { /* Compressed plugin */
    0x0007,0x0001, /*copy 1*/
  0x8050,
  0x0006,0x0002, /*copy 2*/
  0x2a00,0xc000,
  0x0006, 0x801e, 0x0000, /*Rle(30)*/
  0x0006,0x04aa, /*copy 1194*/
  0xf400,0x4095,0x0000,0x02c2,0x6124,0x0024,0x0000,0x0024,
  0x2800,0x1ac5,0x4192,0x4542,0x0000,0x0041,0x2000,0x0015,
  0x0030,0x0317,0x2000,0x0000,0x3f00,0x4024,0x2000,0x0000,
  0x0000,0x0000,0x3e12,0x3800,0x3e00,0xb804,0x0030,0x0015,
  0x0007,0x8257,0x3700,0x984c,0xf224,0x1444,0xf224,0x0024,
  0x0008,0x0002,0x2910,0x0181,0x0000,0x1bc8,0xb428,0x1402,
  0x0000,0x8004,0x2910,0x0195,0x0000,0x1bc8,0xb428,0x0024,
  0x0006,0x0095,0x2800,0x2945,0x3e13,0x780e,0x3e11,0x7803,
  0x3e13,0xf806,0x3e11,0xf801,0x3510,0xb808,0x003f,0xe004,
  0xfec4,0x3800,0x48be,0x17c3,0xfe46,0x41c2,0x48be,0x4497,
  0x4090,0x1c46,0xf06c,0x0024,0x2400,0x2580,0x6090,0x41c3,
  0x6628,0x1c47,0x0000,0x0024,0x2800,0x2449,0xf07e,0x0024,
  0xf400,0x4182,0x673a,0x1c46,0x0000,0x0024,0x2800,0x2589,
  0xf06c,0x0024,0xf400,0x41c3,0x0000,0x0024,0x4224,0x3442,
  0x2900,0xa640,0x4336,0x37c3,0x0000,0x1805,0x2900,0xa640,
  0x4508,0x40c2,0x450a,0x9808,0x0000,0x0207,0xa478,0x1bc0,
  0xc45a,0x1807,0x0030,0x03d5,0x3d01,0x5bc1,0x36f3,0xd806,
  0x3601,0x5803,0x36f3,0x0024,0x36f3,0x580e,0x0007,0x8257,
  0x0000,0x6004,0x3730,0x8024,0xb244,0x1c04,0xd428,0x3c02,
  0x0006,0xc717,0x2800,0x2d05,0x4284,0x0024,0x3613,0x3c02,
  0x0006,0xc357,0x2901,0x6880,0x3e11,0x5c05,0x4284,0x1bc5,
  0x0000,0x0024,0x2800,0x2fc5,0x0000,0x0024,0x3613,0x0024,
  0x3e10,0x3813,0x3e14,0x8024,0x3e04,0x8024,0x2900,0x4f40,
  0x0006,0x02d3,0x36e3,0x0024,0x3009,0x1bd3,0x0007,0x8257,
  0x3700,0x8024,0xf224,0x0024,0x0000,0x0024,0x2800,0x31d1,
  0x3600,0x9844,0x2900,0x3780,0x0000,0x3248,0x2911,0xf140,
  0x0000,0x0024,0x0030,0x0057,0x3700,0x0024,0xf200,0x4595,
  0x0fff,0xfe02,0xa024,0x164c,0x8000,0x17cc,0x3f00,0x0024,
  0x3500,0x0024,0x0021,0x6d82,0xd024,0x44c0,0x0006,0xa402,
  0x2800,0x3695,0xd024,0x0024,0x0000,0x0000,0x2800,0x3695,
  0x000b,0x6d57,0x3009,0x3c00,0x36f0,0x8024,0x36f2,0x1800,
  0x2000,0x0000,0x0000,0x0024,0x3e14,0x7810,0x3e13,0xb80d,
  0x3e13,0xf80a,0x3e10,0xb803,0x3e11,0x3805,0x3e11,0xb807,
  0x3e14,0xf801,0x3e15,0x3815,0x0001,0x000a,0x0006,0xc4d7,
  0xbf8e,0x9c42,0x3e01,0x9c03,0x0006,0xa017,0x0023,0xffd1,
  0x0007,0x8250,0x0fff,0xfd85,0x3001,0x0024,0xa45a,0x4494,
  0x0000,0x0093,0x2800,0x3dd1,0xf25a,0x104c,0x34f3,0x0024,
  0x2800,0x3dd1,0x0000,0x0024,0x3413,0x084c,0x0000,0x0095,
  0x3281,0xf806,0x4091,0x4d64,0x2400,0x4000,0x4efa,0x9c10,
  0xf1eb,0x6061,0xfe55,0x2f66,0x5653,0x4d64,0x48b2,0xa201,
  0x4efa,0xa201,0x36f3,0x3c10,0x36f5,0x1815,0x36f4,0xd801,
  0x36f1,0x9807,0x36f1,0x1805,0x36f0,0x9803,0x36f3,0xd80a,
  0x36f3,0x980d,0x2000,0x0000,0x36f4,0x5810,0x3e12,0xb817,
  0x3e14,0xf812,0x3e01,0xb811,0x0007,0x9717,0x0020,0xffd2,
  0x0030,0x11d1,0x3111,0x8024,0x3704,0xc024,0x3b81,0x8024,
  0x3101,0x8024,0x3b81,0x8024,0x3f04,0xc024,0x2808,0x4800,
  0x36f1,0x9811,0x36f3,0x0024,0x3009,0x3848,0x3e14,0x3811,
  0x3e00,0x0024,0x0000,0x4000,0x0001,0x0010,0x2915,0x94c0,
  0x0001,0xcc11,0x36f0,0x0024,0x2927,0x9e40,0x3604,0x1811,
  0x3613,0x0024,0x3e14,0x3811,0x3e00,0x0024,0x0000,0x4000,
  0x0001,0x0010,0x2915,0x94c0,0x0001,0xcc11,0x36f0,0x0024,
  0x36f4,0x1811,0x3009,0x1808,0x2000,0x0000,0x0000,0x190d,
  0x3600,0x3840,0x3e13,0x780e,0x3e13,0xf808,0x3e00,0x0024,
  0x0000,0x464e,0x0027,0x9e0f,0x2922,0xb680,0x0000,0x190d,
  0x36f3,0x0024,0x36f3,0xd808,0x36f3,0x580e,0x2000,0x0000,
  0x3009,0x1800,0x3613,0x0024,0x3e22,0xb815,0x3e05,0xb814,
  0x3615,0x0024,0x0000,0x800a,0x3e13,0x7801,0x3e10,0xb803,
  0x3e11,0x3805,0x3e11,0xb807,0x3e14,0x3811,0x3e14,0xb813,
  0x3e03,0xf80e,0xb488,0x44d5,0x3543,0x134c,0x34e5,0xc024,
  0x3524,0x8024,0x35a4,0xc024,0x3710,0x8a0c,0x3540,0x4a0c,
  0x3d44,0x8024,0x3a10,0x8024,0x3590,0x0024,0x4010,0x15c1,
[... 471 lines cut ...]
  0x8a0c, 0x5e52, 0x0024, 0x4cb2, 0x4405, 0x0018, 0x0044, 0x654a, /*  340 */
  0x0024, 0x2803, 0xb040, 0x36f4, 0x5810, 0x0007, 0x0001, 0x8e92, /*  348 */
  0x0006, 0x0080, 0x3e12, 0xb814, 0x0000, 0x800a, 0x3e10, 0x3801, /*  350 */
  0x3e10, 0xb803, 0x3e11, 0x7806, 0x3e11, 0xf813, 0x3e13, 0xf80e, /*  358 */
  0x3e13, 0x4024, 0x3e04, 0x7810, 0x449a, 0x0040, 0x0000, 0x0803, /*  360 */
  0x2803, 0xaf04, 0x30f0, 0x4024, 0x0fff, 0xfec2, 0xa020, 0x0024, /*  368 */
  0x0fff, 0xff02, 0xa122, 0x0024, 0x4036, 0x0024, 0x0000, 0x1fc2, /*  370 */
  0xb326, 0x0024, 0x0010, 0x4002, 0x4326, 0x4495, 0x4024, 0x40d2, /*  378 */
  0x0000, 0x0180, 0xa100, 0x4090, 0x0010, 0x0042, 0x4204, 0x0024, /*  380 */
  0xbc82, 0x4091, 0x459a, 0x0024, 0x0000, 0x0054, 0x2803, 0xae04, /*  388 */
  0xbd86, 0x4093, 0x2403, 0xadc5, 0xfe01, 0x5e0c, 0x5c43, 0x5f2d, /*  390 */
  0x5e46, 0x0024, 0x5c56, 0x0024, 0x5e53, 0x5e0c, 0x5c43, 0x5f2d, /*  398 */
  0x5e46, 0x0024, 0x5c56, 0x0024, 0x5e52, 0x0024, 0x4cb2, 0x4405, /*  3a0 */
  0x0010, 0x4004, 0x654a, 0x9810, 0x0000, 0x0144, 0xa54a, 0x1bd1, /*  3a8 */
  0x0006, 0x0413, 0x3301, 0xc444, 0x687e, 0x2005, 0xad76, 0x8445, /*  3b0 */
  0x4ed6, 0x8784, 0x36f3, 0x64c2, 0xac72, 0x8785, 0x4ec2, 0xa443, /*  3b8 */
  0x3009, 0x2440, 0x3009, 0x2741, 0x36f3, 0xd80e, 0x36f1, 0xd813, /*  3c0 */
  0x36f1, 0x5806, 0x36f0, 0x9803, 0x36f0, 0x1801, 0x2000, 0x0000, /*  3c8 */
  0x36f2, 0x9814, 0x0007, 0x0001, 0x8ed2, 0x0006, 0x000e, 0x4c82, /*  3d0 */
  0x0024, 0x0000, 0x0024, 0x2000, 0x0005, 0xf5c2, 0x0024, 0x0000, /*  3d8 */
  0x0980, 0x2000, 0x0000, 0x6010, 0x0024, 0x000a, 0x0001, 0x0d00,
  #define ANALIZER_SIZE 1000
};

    // FINE PATCHES`

fletsche, 2019-01-03:
@MirkoDalmonte Hi Mirko, thank you for sharing your code with the community. However I don't think posting the whole source code into this discussion is maybe not the best way to do that.
I also tried to add some functionality to this project and found it useful to do so by using the tools Github provides for this. Although it took me (and still takes me) some time to understand how it works, I think its worthwhile to read/watch some tutorial and then publish your code via your own (cloned) repository and probably via a pull request.

MirkoDalmonte, 2019-01-03:
Hi, I don't have time to do that, sorry...

koskee, 2019-02-28:
@MirkoDalmonte Ohh excellent! Looks like it was my turn getting caught with my head in the clouds, as I only now noticed your response.. So a big ( and very delayed) thank you. I will give it a try and see how it goes.
Cheers mate :+1:

ememem13, 2019-04-09:
@koskee  Was the spectrum analyzer uploaded?
I get many error messages.

gcharles81, 2019-10-04:
Hello I would like to know if anyone managed to use the pasted spectrum analyzer  code , I cant compile it Thanks

robrob73, 2020-02-16:
Super project ESP Radio&Spectrum analyzer....
It is very sad to waste such great project and don't continue with it. Im not coder, but I very appreciate man, who share his work. Spectrum analyzer very like me but I get many error too.

blotfi, 2020-05-28:
I just adapted and updated the code to have Spectrum analyzer
[https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer](https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer)
Enjoy

MirkoDalmonte, 2020-05-28:
Wowwww!!!
Great job!!!!
Thank you!

Mirko

gcharles81, 2020-05-29:
Hello @blotfi is there a reason why you have not used all the TFT width ? or its because initial code was for a portrait TFT mode

Thanks

blotfi, 2020-05-29:
better ask for this fork in : https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer/issues

blotfi, 2020-05-29:
I also added a video to explain this part:
https://youtu.be/aWUMx9HL5kk

blotfi, 2021-09-01:
check my remote control : 
https://play.google.com/store/apps/details?id=com.embesystems.esp_radio_rc&hl=en_US&gl=US

garangstore, 2025-04-05:
A spectrum analyzer measures the magnitude of an input signal versus frequency within the full frequency range of the instrument. The primary use is to measure the power of the spectrum of known and unknown signals. The input signal that most common spectrum analyzers measure is electrical; however, spectral compositions of other signals, such as acoustic pressure waves and optical light waves, can be considered through the use of an appropriate transducer.[ Spectrum analyzers](https:www.garangstore.com) for other types of signals also exist, such as optical spectrum analyzers which use direct optical techniques such as a monochromator to make measurements.