Edzelf/ESP32-Radio #87: Doesn't save preferences

sellh, 2018-02-13:
Hello! Can you help me in my case? After successful flashing the esp32, I connected to ESP32Radio and I tried to add/change preferences on the first start in the web-interface. I pushed "Save", _but this button didn't work_, I pushed "Restart" - and nothing saved!
Then I cleared a field and wrote only one string: **wifi_00 = XXX/YYY** (Where XXX - ssid of my wi-fi, and YYY - password)._ It is also didn't save!
What I did wrong?

Edzelf, 2018-02-13:
Look into the logging on the serial output.  You should see something like:
```
D: handlehttp started
D: Get command is: saveprefs
D: Send reply for saveprefs
D: writeprefs setstr clk_dst = 1
D: writeprefs setstr clk_offset = 1
D: writeprefs setstr clk_server = pool.ntp.org
D: writeprefs setstr gpio_00 = uppreset = 1
D: writeprefs setstr gpio_12 = upvolume = 2
D: writeprefs setstr gpio_13 = downvolume = 2
D: writeprefs setstr ir_40BF = upvolume = 2
D: writeprefs setstr ir_C03F = downvolume = 2
D: writeprefs setstr mqttbroker = none
D: writeprefs setstr mqttpasswd = *******
D: writeprefs setstr mqttport = 1883
D: writeprefs setstr mqtttpreset = none
D: writeprefs setstr mqttuser = none
.
.
D: End of writing preferences
D: Read 43 keys from NVS
```
I have tested this and found that sometimes is crashed.  It looks like the timer has something to do with it.  Therefore I changed the software to disable the clock during the write-back of the preferences.  See the new version (13-02-2018).

sellh, 2018-02-13:
I upgraded to the new version (13-02-2018). Serial monitor shows an error. Below is a log:
```
D: handlehttp started
D: Get command is: saveprefs
D: Send reply for saveprefs
D: writeprefs setstr gpio_00 = uppreset = 1
D: writeprefs setstr gpio_12 = upvolume = 2
D: writeprefs setstr gpio_13 = downvolume = 2
D: writeprefs setstr gpio_14 = stop
D: writeprefs setstr gpio_17 = resume
D: writeprefs setstr gpio_21 = station = icecast.omroep.nl:80/radio1-bb-mp3
D: writeprefs setstr mqttbroker = none
D: writeprefs setstr mqttport = 1883
D: writeprefs setstr mqttuser = none
D: writeprefs setstr mqttpasswd = *******
D: writeprefs setstr mqqprefix = none
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d8497  PS      : 0x00060630  A0      : 0x800da145  A1      : 0x3ffd6950  
A2      : 0x00000000  A3      : 0x3f40162c  A4      : 0x00000000  A5      : 0x3ffc70e0  
A6      : 0x00000000  A7      : 0x00000004  A8      : 0x800d8368  A9      : 0x3ffd6930  
A10     : 0x000000c0  A11     : 0x3ffd6968  A12     : 0x00000005  A13     : 0x3ffe2580  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d8497:0x3ffd6950 0x400da142:0x3ffd6a90 0x400da3d3:0x3ffd6b70 0x4012bbf4:0x3ffd6bb0

Rebooting...
```

Edzelf, 2018-02-13:
Mqqprefix must be a typing error in the default config. Try to remove this line.

sellh, 2018-02-13:
I deleted _Mqqprefix_, but nothing changed. Then I found problem line! If delete line with wifi settings, config will save quite normal. 
```
D: handlehttp started
D: Get command is: saveprefs
D: Send reply for saveprefs
D: writeprefs setstr gpio_00 = uppreset = 1
D: writeprefs setstr gpio_12 = upvolume = 2
D: writeprefs setstr gpio_13 = downvolume = 2
D: writeprefs setstr gpio_14 = stop
D: writeprefs setstr gpio_17 = resume
D: writeprefs setstr gpio_21 = station = icecast.omroep.nl:80/radio1-bb-mp3
D: writeprefs setstr mqttbroker = none
D: writeprefs setstr mqttport = 1883
D: writeprefs setstr mqttuser = none
D: writeprefs setstr mqttpasswd = *******
D: writeprefs setstr mqqprefix = none
D: writeprefs setstr volume = 72
D: writeprefs setstr toneha = 0
D: writeprefs setstr tonehf = 0
D: writeprefs setstr tonela = 0
D: writeprefs setstr tonelf = 0
D: writeprefs setstr preset = 6
D: writeprefs setstr preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
D: writeprefs setstr preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
D: writeprefs setstr preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIAMI 256k
D: writeprefs setstr preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
D: writeprefs setstr preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
D: writeprefs setstr preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
D: writeprefs setstr preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
D: writeprefs setstr preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
D: writeprefs setstr preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
D: writeprefs setstr preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
D: writeprefs setstr preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
D: writeprefs setstr preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
D: writeprefs setstr clk_server = pool.ntp.org                            # Time server to be used
D: writeprefs setstr clk_offset = 1                                       # Offset with respect to UTC in hours
D: writeprefs setstr clk_dst = 1                                          # Offset during daylight saving time (hours)
D: writeprefs setstr ir_40BF = upvolume = 2
D: writeprefs setstr ir_C03F = downvolume = 2
D: writeprefs setstr pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
D: writeprefs setstr pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
D: writeprefs setstr pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
D: writeprefs setstr pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
D: writeprefs setstr pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
D: writeprefs setstr pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
D: writeprefs setstr sd_cs = 21                                           # GPIO Pin number for SD card "CS"
D: writeprefs setstr pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
D: writeprefs setstr pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
D: writeprefs setstr pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
D: End of writing preferences
D: Read 44 keys from NVS
```

koskee, 2018-02-14:
Haha. Are your router credentials really "xxx/yyy"? If so, that could cause problems.  You could try renaming your WiFi (you'd probably want something more secure than that anyway). 
If that isn't possible, then try running a search in the code for `wifi_xx` and changing the `_xx` to a different letter (maybe don't choose 'y' either, obviously lol) 

@Edzelf - I suspect it has something to do with the `String readprefs ( bool output )` function. Where it is searching for `wifi_xx`,  it seems to cause the ESP to read from an undefined pointer/ out-of-bounds memory (judging based on the load prohibited panic,  that is)

koskee, 2018-02-14:
@Edzelf - I solved the timer issue a while back in my personal version, by simply adding a boolean parameter that is passed into the function like `void handleSaveReq (bool force)` which then checks this value and immediately performs the write if true. Then I can use it whenever I need to be sure the timer doesn't interfere - like when a software reboot is called, or when writing prefs, etc. 
(I haven't had a look at your changes, but I found in testing that they still might not be saved if something changes and then a reboot is triggered right after)

sellh, 2018-02-14:
Lol :) **@koskee** - of course I set normal ssid and password of my wi-fi at my router. **xxx/yyy** is just an example.
But how to solve the problem with `wifi_00` saving I do not know.

Edzelf, 2018-02-14:
Can you supply an example for "wifi_00=" that fails, so I can test it.

sellh, 2018-02-14:
I create a few wi-fi networks, but nothing helped. Below are three examples:
**wifi_00 = IGOR_wifi/(alarm)**
or
**wifi_00 = test/27031984**
of
**wifi_00 = ITG/alarm** 

Maybe is it problem with the board (doit esp32 devkit v1)? Because last year I used your project of wifi radio on 8266, and it worked fine.

Edzelf, 2018-02-15:
Yes, there was a bug in saving the WiFi credentials.  The version of 15-feb-2018 should work now.

sellh, 2018-02-15:
**@Edzelf** Thank's a lot! Now it works fine.

closed 2018-02-15 by sellh