Edzelf/ESP32-Radio #88: error: 'adc1_get_raw' was not declared in this scope

djuscha, 2018-03-03:
wanted to try a new version and get now when compiling the error message:(

C:\Users\Andreas\Desktop\ESP32-Radio-master\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'void timer100()':

Esp32_radio:1700: error: 'adc1_get_raw' was not declared in this scope

   adcval = ( 15 * adcval +  adc1_get_raw ( ADC1_CHANNEL_0 ) ) / 16 ;

                                                           ^

exit status 1
'adc1_get_raw' was not declared in this scope

Edzelf, 2018-03-03:
This function is deprecated and possibly not avalable in new libraries.  Try to remove "ADC_CHANNEL0".  The line should read now:
adcval = ( 15 * adcval + adc1_get_raw( ) ) / 16 ;
The new function does not need the parameter.

djuscha, 2018-03-03:
It does not work that way.
adcval = ( 15 * adcval + adc1_get_raw( ) ) / 16 ;

but I have the line out and go now :)
// adcval = ( 15 * adcval +  adc1_get_raw ( ADC1_CHANNEL_0 ) ) / 16 ;

Now I have only the problem that the IR remote control does not work
no reaction in debug output ?!

Edzelf, 2018-03-03:
There was a bug in determining the GPIO number of the IR.  It is corrected in the new version I just uploaded.

djuscha, 2018-03-03:
Thank you very much!!! Now go !! ;)

vadpol, 2018-03-04:
I had the same problem with adc1_get_raw until I've upgraded the xtensa-esp32-elf environment which also includes all the relevant libraries to the freshest one xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip available at github. Also I was forced to change GPIO 36 (ADC1_CHANNEL_0) to GPIO 33 (ADC1_CHANNEL_5) since GPIO 36 was not broken up in my compact ESP32 adapter pcb. Now I have the blue/red (also changed green to blue for more colorful screen) battery capacity bar under the top text line on display. My parameters in config screen are: 
bat0 = 2218
bat100 = 2718
The resistors in voltage divider are the same as recommended by Ed: 22k and 100k.

Also I encountered the problem that IR stopped working in previous software version, no reaction in debug as well. I will check out the new one.

Edzelf, 2018-03-05:
adc1_get_raw is defined in this [library](https://github.com/espressif/arduino-esp32).

vadpol, 2018-03-05:
IR works again, just checked, but there is another issue appeared with the version from 15, February and still continues with version of 3, March - VS1053 gives the error in debug while SD module is connected, consequently no sound. However the display and the rest work ok. Without SD module connected no problem, VS1053 works correctly. Also no problem with code version from 2, January. I will open the new issue with my debug log, what could be the reason in this case?

stepner03, 2018-06-04:
Sorry ...I installed last version xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip, but ESP32_test_ADC:58: error: 'adc1_get_raw' was not declared in this scope

     val += adc1_get_raw(ADC1_CHANNEL_6); 

                                       ^
exit status 1
'adc1_get_raw' was not declared in this scope

How is it possible ?

Edzelf, 2018-06-04:
I am a Windows user and I used get.exe in ...\Documents\Arduino\hardware\espressif\esp32\tools to install the toolchain.  That gives me:
```
System: Windows, Info: Windows-10-10.0.17134
Platform: i686-mingw32
Tool xtensa-esp32-elf-win32-1.22.0-75-gbaf03c2-5.2.0.zip already downloaded
Extracting xtensa-esp32-elf-win32-1.22.0-75-gbaf03c2-5.2.0.zip

```
In ...\Documents\Arduino\hardware\espressif\esp32\tools\sdk\include\driver\driver\adc.h you will find the definition for adc1_get_raw:
```
/**
 * @brief Take an ADC1 reading on a single channel
 *
 * @note Call adc1_config_width() before the first time this
 * function is called.
 *
 * @note For a given channel, adc1_config_channel_atten(channel)
 * must be called before the first time this function is called.
 *
 * @param  channel ADC1 channel to read
 *
 * @return
 *     - -1: Parameter error
 *     -  Other: ADC1 channel reading.
 */
int adc1_get_raw(adc1_channel_t channel);

```
So, are you sure that the right header is included?  And what is "ESP32_test_ADC"?

stepner03, 2018-06-04:
Thank you ...now is OK !!! 
But I have another error 

rchiving built core (caching) in: G:\Users\HOME1~1\AppData\Local\Temp\arduino_cache_757160\core\core_espressif_esp32_esp32_FlashFreq_80,UploadSpeed_115200,DebugLevel_none_898f9e38ebae302494d0e119be95335d.a
usage: gen_esp32part.exe [-h] [--disable-md5sum] [--verify] [--quiet]

                         input [output]

gen_esp32part.exe: error: argument input: can't open 'C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/partitions/{build.partitions}.csv': [Errno 2] No such file or directory: 'C:\\Program Files (x86)\\Arduino\\hardware\\espressif\\esp32/tools/partitions/{build.partitions}.csv'

exit status 2 

Does anyone have an idea?