Edzelf/ESP32-Radio #89: SD module triggers VS1053 error, no sound consequently

vadpol, 2018-03-05:
Hi Ed,

There is no sound in any mode (radio or mp3 player) with the code versions as from 15, February and 3, March due to VS1053 error while SD card module is connected. No problem when SD card module is disconnected. Could you please elaborate on it? 

Here is the debug log:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sat, 03 Mar 2018 15:44:00 GMT.  Free memory 171836
D: Partition nvs found, 20480 bytes
D: Read 50 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: Enable pin 35 for IR
D: Start TFT
D: Locate mp3 files on SD, may take a while...
D: 45 tracks on SD
D: Create list with acceptable WiFi networks
D: Added Cisco1200 to list of networks
D: Added AndroidAP to list of networks
D: Added Faraway to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 6
D:  1 - Faraway                   Signal: -55 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - Kalinins                  Signal: -82 dBm, Encryption WPA_WPA2_PSK, 
D:  3 - Sova1126                  Signal: -83 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - Cisco1200                 Signal: -83 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  5 - Lolinda                   Signal: -88 dBm, Encryption WPA_WPA2_PSK, 
D:  6 - Riahhaar                  Signal: -92 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: bat0 with parameter 2318
D: Command: bat100 with parameter 2818
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 2
D: Command: clk_server with parameter pool.ntp.org
D: Command: ir_02fd with parameter mp3track = 0
D: Command: ir_22dd with parameter downpreset = 1
D: Command: ir_629d with parameter upvolume = 2
D: Command: ir_a857 with parameter downvolume = 2
D: Command: ir_c23d with parameter uppreset = 1
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_shutdown with parameter -1
D: Command: pin_spi_miso with parameter 19
D: Command: pin_spi_mosi with parameter 23
D: Command: pin_spi_sck with parameter 18
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 8
D: Command: preset_00 with parameter sc-atr.1.fm:7700
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter air.radiorecord.ru:8102/sd90_320
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter listen1.myradio24.com:9000/5967
D: Command: preset_05 with parameter cast.radiogroup.com.ua:8000/avtoradio
D: Command: preset_06 with parameter air.radiorecord.ru:805/sd90_320
D: Command: preset_07 with parameter cast.radiogroup.com.ua:8000/retro
D: Command: preset_08 with parameter nostalgia.fm:8000/nostalgiafm.mp3
D: Command: preset_09 with parameter cast.radiopyatnica.com.ua/radiopyatnica
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 4
D: Command: tonehf with parameter 4
D: Command: tonela with parameter 8
D: Command: tonelf with parameter 7
D: Command: volume with parameter 86
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF
D: VS1053 error retry SB:012C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0258 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0384 R1:FFFF R2:FFFF
D: VS1053 error retry SB:04B0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:05DC R1:FFFF R2:FFFF
D: VS1053 error retry SB:0708 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0834 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0960 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0A8C R1:FFFF R2:FFFF
D: VS1053 error retry SB:0BB8 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0CE4 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0E10 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0F3C R1:FFFF R2:FFFF
D: VS1053 error retry SB:1068 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1194 R1:FFFF R2:FFFF
D: VS1053 error retry SB:12C0 R1:FFFF R2:FFFF
D: VS1053 error retry SB:13EC R1:FFFF R2:FFFF
D: VS1053 error retry SB:1518 R1:FFFF R2:FFFF
D: VS1053 error retry SB:1644 R1:FFFF R2:FFFF
D: Connect to WiFi
D: Connected to Faraway
D: IP = 192.168.1.125
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 21:27:13
D: STOP requested
D: New preset/file requested (8/0) from nostalgia.fm:8000/nostalgiafm.mp3
D: Connect to new host nostalgia.fm:8000/nostalgiafm.mp3
D: Connect to nostalgia.fm on port 8000, extension /nostalgiafm.mp3
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:128
D: Headerline: icy-genre:Retro
D: Headerline: icy-name:NostalgiaFM
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://nostalgia.fm
D: Headerline: Server: Icecast 2.4.0-kh7
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -  FFFF
D:   1 -  FFFF
D:   2 -  FFFF
D:   3 -  FFFF
D:   4 -  FFFF
D:   5 -  FFFF
D:   6 -  FFFF
D:   7 -  FFFF
D:   8 -  FFFF
D:   9 -  FFFF
D:   A -  FFFF
D:   B -  FFFF
D:   C -  FFFF
D:   D -  FFFF
D:   E -  FFFF
D:   F -  FFFF
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Playtask
CPU 1: IDLE
D: Metadata block 48 bytes
D: Streamtitle found, 41 bytes
D: StreamTitle='Leonard Cohen - The Future';
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Playtask
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Playtask
CPU 1: IDLE
D: Metadata block 32 bytes
D: Streamtitle found, 30 bytes
D: StreamTitle='Sting - Roxanne';

Edzelf, 2018-03-05:
It looks like the SD card spoils the SPI signals.  Do you have the oppertunity to meassure the signals?  Otherwise, try 10 k pull-ups for MOSI, MISO and CLK.

vadpol, 2018-03-05:
I suspect something like this however it was working good in before. I will add the pull up resistors to see if it helps. I will report it herewith.

Thanks for you fast reaction Ed!

vadpol, 2018-03-05:
Pulling up doesn't help so far. Next I will try to hook up an oscilloscope to SPI bus.

vadpol, 2018-03-12:
I have broken my brain already - pulled up SPI bus with resistors, checked all the wiring on correctness, short cuts and lose wire, changed SD card module to a brand new just out of the box - nothing helps. It doesn't matter whether SD card installed in the slot or not. As soon as SD module is not connected everything works fine, as soon as I connect it everything works until SPI tests are started with VS1053 - there are all "1" in its all registers and consequently no sound... I can't imagine at the moment what the reason it could be....

locki-cz, 2018-03-12:
I get idea ... there can be some HW problem on slot for SD card? Please check if it is ok. If it worked before, can you test old sketch or some different SD slot?

vadpol, 2018-03-12:
Checked in before - it doesn't work with old software as well (previously worked). Slot is OK, checked it with two brand new modules - all the same...

vadpol, 2018-03-19:
Finally I've got an oscilloscope and it helped me to figure out the reason of the problem. It lies in the tricky schematics of some Chinese SD modules. The one I have is made by Catalex and the wrong thing with these modules is that MISO signal produced by the module was constantly blocking the SPI bus due to level shifter LV125A chip control. Actually a MISO signal from the module has to switch off (z-state) while CS card signal is high per schematics below, but this was not the case. Therefore SPI bus was always occupied by the SD module when connected which blocked all the data exchange in between ESP32 and VS1053. To solve the problem some correction of the module schematics has to be done as per attached picture:

[image]

The MISO output signal has to be connected directly to SD card interface (dataline DO) without involvement of the level shifter which created a problem. In order to do so the MISO signal track leading to the pin 11 of LVC125A has to be cut and connected with a small wire jumper to the outer pad of the previously removed R4 resistor and MISO pin at the module's header. See it at the pcb layout in the picture below.

[image]
 
This is quite a tiny job however it is still possible with standard soldering iron, see how it looks in my case. 

[image]

I have to admit that this has fully solved my problem which appeared to be a pure HW indeed. However I still wonder how it used to work previously for a certain period without this upgrade. So pay attention to the schematics of the modules before connecting them, it could be really tricky!

Edzelf, 2018-03-19:
Great work.  But one thing I do not understand.  The 74HC125 IC should put MISO in a high-impedance state when CS is HIGH.....  And CS should be high if the SD card is not selected, which is the normal operation mode.

vadpol, 2018-03-19:
Hi Ed, exactly, that's what I expected as well: when CS is high the module's MISO should be disconnected, but... it was not the case. I digged down into the issue and found out that these Catalex modules could be different in schematics - at some of them the 'CS card' signal is connected to the general CS signal coming from ESP32 via level shifter as per my schematic diagram in above (I used not the exact schematics here, see below the right one, sorry). I found first the one in above and thought that it is all like that. But later I found that at some modules it could be different and connected to the ground like this:

[image]

which was exactly my case. In such a case the module's MISO signal is always active and interferes the SPI bus. Therefore I said that schematics is tricky - you never know what you get in reality.

Edzelf, 2018-03-19:
You are right.  I have a module here like yours, not using it for the radio though.  Pin 13 of the LVC125A is connected to GND indeed.

vadpol, 2018-03-19:
Another solution would be to connect pin 8 and pin 13 of LVC125A with wire jumper as well, but first pin 13 should be disconnected from GND by cutting off the relevant track.

closed 2018-03-19 by vadpol

Edzelf, 2018-03-20:
I agree.  That would be the best modification.