Edzelf/ESP32-Radio #90: IR interface feature not working

smeckcems, 2018-03-08:
Hi,
I have succesfully set-up the radio (thanks for great work!), music plays, config works... but I am not able to make IR interface available. 
Those are parts from debug related to IR:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 171972
D: Partition nvs found, 20480 bytes
D: Read 49 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
...
...
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start TFT
D: Create list with acceptable WiFi networks
...
...
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_ir with parameter 35
D: Command: pin_spi_miso with parameter 19
D: Command: pin_spi_mosi with parameter 23
...
...

So I think that all is configured and connected properly. Problem is, that I can't see in debug any message like "D: IR code 807F received, but not found in preferences!" after pressing the button on IR commander. IR receiver VS1838 is working well, because I see pulses on GPIO35 with my oscilloscope.

Is there anything else to set to make IR working? 
Thanks

Edzelf, 2018-03-08:
No, it should work like this.  Check GPIO35 once again, maybe the wrong pin, test it by making it low and do a restart: all the pin levels are printed for debugging.
Try a different remote.

smeckcems, 2018-03-09:
Hmm, so that something is still wrong...
GPIO35 looks working good, grounded pin says D: GPIO35 is LOW, probably no PULL-UP
Other pins show the same behavior - no IR receive, but grounding is ok (tried pins 22, 26, 32, 33...) also with different commander and VS1838.

Edzelf, 2018-03-09:
Sorry, I cannot reproduce the error here.  Just tested it again.
It looks like the interrupt on GPIO35 does not work in your case.

smeckcems, 2018-03-09:
Interrupt is working, software is jumping into isr_IR() routine when I press button on commander, but: 
1. returned "ir_value" is always zero,
2. value "intval" is changing during receiving command from commander, so something is happening here.
I will continue in exploration later...

smeckcems, 2018-03-09:
At routine void IRAM_ATTR isr_IR() I had to change intervals to recognize pulses:
if ( ( intval > 300 ) && ( intval < 950 ) )        // Short pulse?
...
else if ( ( intval > 1500 ) && ( intval < 1950 ) ) // Long pulse?

And then set up correct level changes:
else if ( ir_loccount == 23 )                      // Value is correct after 65 level changes

to have at least some received commands (however I see there some unstability in received patterns, so wrong commands are interpreted). But this looks very specific for each remote controller. For example I have this one:
https://www.amazon.com/HDE-Universal-Stealth-Television-Keychain/dp/B0092TH8WS

I have tested also some old Sony remote and there ir_loccount have to be set to 8 and for Panasonic it is completely not usable - some keys are available only for ir_loccount == 43, but some require 45 or 47...

So now only stability of recognized command is issue...

Edzelf, 2018-03-10:
Samsung works, LG works, and I have 3 cheap chinese remotes like [this](https://www.aliexpress.com/item/HX1838-VS1838-Infrared-IR-Wireless-Remote-Control-Sensor-Module-Kits/32639412147.html?spm=2114.search0104.8.24.68ef26cdYXf7oy&priceBeautifyAB=0) that all work.

smeckcems, 2018-03-12:
Ok, thanks. I will order the same type and let you know.

vadpol, 2018-03-12:
@smeckcems check the setup routine in the code you currently uploaded to the ESP32, maybe you still have a bug with ir pin like in the code version as from 15, February. The symptom was exactly the same - no reaction on any IR command, no respond in debug log. Despite it was fixed by Ed in the next code versions you could still have it for whatever reason (copy/paste issue, mixing up of the versions, etc).

smeckcems, 2018-03-13:
thanks, checked, I have latest version from 5.3.2018, but issue is still persisting...

Edzelf, 2018-03-13:
I ordered a keyring remote, to do some testing.

smeckcems, 2018-03-23:
I'm back, ordered remote has arrived. I have returned back code changes to original and checked [this](https://www.aliexpress.com/item/HX1838-VS1838-Infrared-IR-Wireless-Remote-Control-Sensor-Module-Kits/32639412147.html?spm=2114.search0104.8.24.68ef26cdYXf7oy&priceBeautifyAB=0) type of remote - and it work perfectly.

Edzelf, 2018-03-23:
Great!

Edzelf, 2018-04-12:
The Chinese keyring remote arrived.  It works like a charm....