Edzelf/ESP32-Radio #91: Esp32_radio.ino ::::  shutdown_pin >= 0 ::::

pippo-75, 2018-03-09:
Goodmorning,
here following there are some code excerpts from his project * Esp32_radio.ino *
There are 3 code line where < if ( shutdown_pin >= 0 ) > has been used:

void VS1053::begin() {
  .........................................................................
  if ( shutdown_pin >= 0 )                             // Shutdown in use?
  {
    pinMode ( shutdown_pin,   OUTPUT ) ;
    digitalWrite ( shutdown_pin, HIGH ) ;              // Shut down audio output
  }
  .........................................................................
    
void VS1053::startSong() {
  sdi_send_fillers ( 10 ) ;
  if ( shutdown_pin >= 0 )                             // Shutdown in use?
  {
    digitalWrite ( shutdown_pin, LOW ) ;               // Enable audio output
  }
}

void VS1053::stopSong() {
  .........................................................................
  sdi_send_fillers ( 2052 ) ;
  if ( shutdown_pin >= 0 )                             // Shutdown in use?
  {
    digitalWrite ( shutdown_pin, HIGH ) ;              // Disable audio output
  }
  .........................................................................
  
Now, if I understand correctly, shutdown_pin (as stated) is a Digital Pin
(VS1053::begin), so its value can be read as HIGH / LOW (1/0) using 
digitalRead or reading the relative "bit" somewhere in the register.
Also, in any case, if we comparing a 'value' simply if it's -Equal to- or 
-Higher than- 0 (zero), as result we always get  >>true<< 

Please correct me if I'm wrong, but I think it's a typo.

Anyway, thanks for Your time and this great project.

Edzelf, 2018-03-09:
No, shutdown_pin is a variable defining the GPIO to be used for shutdown of the amplifier.
A negative value is assigned to it if the shutdown function is not used.
Shutdown is an output bit, so its value is never read by digitalRead().

pippo-75, 2018-03-09:
Hello, obviously I'm wrong: I'm sorry.
Here following an excerpt where >> shutdown_pin << is declared:

//**************************************************************************************************
// VS1053 class definition.                                                                        *
//**************************************************************************************************
class VS1053
{
  private:
    uint8_t       cs_pin ;                        // Pin where CS line is connected
    uint8_t       dcs_pin ;                       // Pin where DCS line is connected
    uint8_t       dreq_pin ;                      // Pin where DREQ line is connected
    **_uint8_t       shutdown_pin ;                  // Pin where the shutdown line is connected_**
    uint8_t       curvol ;                        // Current volume setting 0..100%
    const uint8_t vs1053_chunk_size = 32 ;
    // SCI Register

The thing that I don't understand is: it's really declared as unsigned or, instead,
I've to look somewhere in the code ?

Please, excuse me for my insistence and have a good day.

Edzelf, 2018-03-09:
Yes, it should be declared as a int8_t.  I will fix it.