Edzelf/ESP32-Radio #93: USE TFT - howto

njordan77, 2018-03-26:
Hi, is there a Define for not using TFT. I would like to use it headless. Is there any flag like in the ESP8266 version, could not see it. What would be the easiest way to get rid of the TFT parts of the code, or do i have to shrink it from removing library until there is no compile error anymore?

Also, what if MQTT is not used, can i somehow deactivate it so that it does not wait for a non existing broker?

Regards Norbert

Edzelf, 2018-03-26:
From my memory: for TFT, define pin number -1 for the pins and let the software in.  For MQTT, define "none" for the broker.
I will check these settings later (it's midnight here).

Edzelf, 2018-03-27:
I've checked it.  Just define tft_cs_pin as "-1" and mqttbroker as "none".  Leave the software as it is, do not mind the space wasted for unused functions.

njordan77, 2018-03-27:
Thanks a lot, will try today.
Sidequestion....i have a 3,5inch ILI9486 display. from your code i see its a different ILI display supported. Will i have a chance using the same pins that my chip is compatible, any idea? ever tried various types? I normally work with ILI9341 which is widely used, but my new one and your main supported are not so well known to me...so i'm questioning if its stupid to believe that there is kind of compatibility.

Edzelf, 2018-03-27:
I don't think the two are compatible.  But it shouldn't be too hard to change the code if necessary.  Search for the lines that have "tft->" in it.

njordan77, 2018-03-27:
sorry - one more question. As i'm headless with webserver only. I guess SD Card is useless this way (in addition to not having any SD-holder, is there a way to disable CD-card as well.....the single SD-pin to -1 as well? Thanks

Edzelf, 2018-03-27:
Yes, set sd_cs_pin to -1.

njordan77, 2018-03-27:
hm, any idea what goes wrong on an Sparfun ESP32thing....vs1053 only connected.

```
ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 171740
D: Partition nvs found, 20480 bytes
D: Read 0 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40085fc5  PS      : 0x00060033  A0      : 0x4008744b  A1      : 0x3ffd32c0  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc8494  A5      : 0x3ffc8654  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000016  A9      : 0x00000002  
A10     : 0x3ffc11c0  A11     : 0x00000000  A12     : 0x3ffc8654  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085fc5:0x3ffd32c0 0x40087448:0x3ffd32e0 0x400873fe:0x00000000

Rebooting...
```

njordan77, 2018-03-27:
hm, any idea what goes wrong. Sparkfun ESP32thing, VS1053 only connected.....looping all the time....

ts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 171740
D: Partition nvs found, 20480 bytes
D: Read 0 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40085fc5  PS      : 0x00060033  A0      : 0x4008744b  A1      : 0x3ffd32c0  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc8494  A5      : 0x3ffc8654  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000016  A9      : 0x00000002  
A10     : 0x3ffc11c0  A11     : 0x00000000  A12     : 0x3ffc8654  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40085fc5:0x3ffd32c0 0x40087448:0x3ffd32e0 0x400873fe:0x00000000

Rebooting...

Edzelf, 2018-03-27:
Make sure you have the latest core libraries.

njordan77, 2018-03-27:
if you mean ESP32 core, mine is 2 weeks old. will update if at all a new snapshot is available. but fear this is something different making it loop all the time....hm, thought you may have seen similar stuff before....

naitsimp, 2018-04-05:
@njordan77 Did you solve your problem, because I have the same?

jgrulich, 2018-04-08:
Use another, more stronger power supply, at least 5V/1A, or at least 3300uF capacitor at the Vin and GND pins of the ESP module. In my case it worked.

kosso, 2018-05-29:
I'm also getting the 'Guru Meditation loop', just after the OTA and mDNS responder starts. 

Running on the ESP32-WROVER-KIT V3 with on board TFT. (Though only defining the 'dummytft' for now)

Edzelf, 2018-05-30:
The 30-may function has a bugfix.  Global variables used in ISRs did not have the DRAM attribute. 
If you have "Guru meditation" errors, please put the output of use the "Exception Decoder" in the IDE tools in your comments.  That will help to locate the problem.

kosso, 2018-05-30:
OK. Just updated to the latest version today. 
And using the Exception Decoder tool.

Here's one of my Guru Meditation Errors and the decoded stack. 
Interestingly, it's still referring to someone else's code paths?? 

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4012b46e  PS      : 0x00060f30  A0      : 0x800f00c4  A1      : 0x3ffe3bc0  
A2      : 0x0000001a  A3      : 0x3ffc8ffc  A4      : 0x3ffc1a00  A5      : 0xcccccccc  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x8012b46b  A9      : 0x3ffe3ba0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xcccccccc  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4012b46e:0x3ffe3bc0 0x400f00c1:0x3ffe3be0 0x40082dea:0x3ffe3c00 0x40083060:0x3ffe3c30 0x400788a6:0x3ffe3c50 0x4007890d:0x3ffe3c80 0x40078a52:0x3ffe3cb0 0x40078b7a:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
...
```

decoded: 

```
PC: 0x4012b46e: periph_module_enable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/include/soc/dport_access.h line 102
EXCVADDR: 0xcccccccc

Decoding stack results
0x4012b46e: periph_module_enable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/include/soc/dport_access.h line 102
0x400f00c1: esp_perip_clk_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./clk.c line 271
0x40082dea: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 283
0x40083060: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 203

```

I have no idea who ` /Users/ficeto/Desktop/` is ;)

Edzelf, 2018-05-31:
The 31-may version will not crash anymore.  The reason was that I2C was used but no pins were defined for sda and scl.  Now the radio will start (although without a working display) and you can use the config page in the webinterface to add the pinnumber for I2C, for example:
```
pin_tft_scl = 13                              # GPIO Pin number for SCL
pin_tft_sda = 14                              # GPIO Pin number for SDA
```

kosso, 2018-05-31:
Thanks. I'll give this a try. 

By the way, I still don't understand why the decoded stack trace shows someone else's toolchain path. (I can see from a Google search that it appears in other projects too. Who is this elusive 'ficeto' user? ;) )

Edzelf, 2018-05-31:
It's the directory of the sourcefile at the moment the sourcefile was compiled by Espressif.  The library is an object library, the source is not distributed.