Edzelf/ESP32-Radio #94: Download MP3 to IFFS and play from there

udavst, 2018-03-29:
I want to make a doorbell and other reminders.
How to download short mp3 in IFFS (esp32), and play on MQTT command?

Edzelf, 2018-03-29:
I do have an ESP32 sketch that can play ogg-files from SPIFFS.  The soundfiles are decoded by software and send to the 2 DACs of the ESP32.  Maybe you can add the MQTT interface.  If you are interested, I can put it on github.

udavst, 2018-03-29:
I would like to use the capabilities of the built-in player in esp32-radio. Just specify as an argument not a network or SD path, but a path to internal memory. Or is it impossible?

udavst, 2018-03-29:
Help please an example, I'm not a programmer.

Edzelf, 2018-03-30:
Adding an SD card  may cost you 6 dollar.  Adding SPIFFS software and testing may cost 3 hours.  That's much more effort....