Edzelf/ESP32-Radio #97: TFT - Sidebar and filling numbrs

JocB1, 2018-04-09:
On the left side of the TFT I have a bar with about 1/6 width. 
First I have a good display, then the TFT is filled by numbers like "10369" till the area is compleately filled. 

In the PDF there are not for all librarys links. Is there a full list for the librarys to usw. Maybe I habe a wrong one. But the Library for TFT is the one linked in the PDF. I´ve done the TFT library workouround.

Edzelf, 2018-04-09:
See issue 56 for a complete list of libraries used.

JocB1, 2018-04-09:
The libraries seems to be ok. Could it be a problem with the kind of ESP-Moudule: I use a bare "ESP-WROOM-32" from Aliexpress (Arduino says: "Chip is ESP32D0WDQ6 (revision 1)" and "Auto-detected Flash size: 4MB". I use the Arduino setting "ESP 32 dev Module", but it works only, when i chose a "Debug Level", for example "verbose", when I deselect the debug level,  the ESP is restarting in a loop. 
So I´m not sure which is the wright board setting in Arduino for uploading.

JocB1, 2018-04-10:
I will check it with a real 5V Power supply instead of the Notebooks USB-Port. 

My TFT uses a ST7735C chip. Maye it´s the problem, that this one ist not 100% compatible to the ILI9xxx ??

Edzelf, 2018-04-10:
The ST7735C if a different chip.  You have to use the right library for this, and maybe change som code that address the display.   There are only a few lines concerning the display.

JocB1, 2018-04-11:
YES, with the changed TFT code, which I found from "derpicknicker1" in the pull requests it works now. 
Is it possible to take the conditional code from "derpicknicker1" to switch between the 2 types of TFT per Define?