Edzelf/ESP32-Radio #98: How to run this code on esp32 at 160MHz

mottobug, 2018-04-10:
Hello,

first of all thanks for this nice piece of code. I've build this project and it's working nice with the exception of the user interface is a little slow. I'm wondering if i'm running this at the correct speed. In the PDF documentation its running at 160MHz. I'm only able to select 40 or 80MHz at the Arduino IDE. 

Am i missing something here? 

Sincerely, Thomas

Edzelf, 2018-04-10:
I will check this later, but I think the speed in the IDE is not the CPU speed.  It could be the flash memory speed.

koskee, 2018-04-11:
Yes, the 40/80 choice is for the flash memory. As far as I know it should be set to 40Mhz... The 80 option wasn't working properly for either some boards or all boards, I can't remember which. 
This is just off the top of my head based on something I read at some point however, and it may have been fixed already at some point so feel free to correct me if you know something different. 

The CPU should be running at 240Mhz by default and I don't think is changeable without digging into modifying register bits (or using the make menuconfig possibly?), as it isn't implemented in any of the arduino core thus far as far as I've seen - and I've been sifting through a bunch of core files pretty recently 

Are you using the latest version of both software and libraries? It may be worth checking if you haven't already .

mottobug, 2018-04-13:
You're right, its the flash memory speed. I'll try and set it to 40MHz. I'm referring to the response to the rotary encoder, which is the only interface to the radio i know of (exception web-interface). When i press the button once it takes up to 2 seconds to mute (mostly about one second). Two and three presses to get to the station or track menu only works if i try it several times. When i'm in the menu scrolling with the encoder ist slow and unpredictable. Playback works fine. The Clock sometimes skips seconds. 

I'm going to check of actual library versions at the weekend. Arduino IDE ist latest available.

koskee, 2018-04-16:
Honestly, I haven't had much luck with the rotary encoder interface either. I believe it is highly dependent on the specific characteristics of the hardware you are using:

I have two seemingly identical encoders but in trying to use each of them, i'm finding that they are very different internally. 
For example, a clockwise rotation on one will change a parameter upwards, while rotation in the same direction on the other encoder will produce a downward change. This is in addition to some of the timing problems you have reported. 

I had found some code a while back that gives more reliable movement on one of them, but for the other to function as expected this new code had to be modified quite a bit before I had it working. Since it uses software debouncing constants as well as some additional discrete physical components, I don't feel I can suggest it as a more appropriate solution due to the amount of trial and error involved with getting it to work (there could be an infinite number of similar but slightly different switches out there that would require the code to be adjusted specifically for it) 

I think the code used now in the project is likely to at least be marginally functional in most cases, with most switches, since it does not rely on debouncing. It should work for the majority of users without requiring an oscilloscope to measure & properly characterize the parameters of each switch, and should be generally functional "out of the box", although seemingly not without its quirks. 

You might like to try another encoder to see if you have better results, or if you are feeling ambitious, try reading into some of the other issues posted, this has come up before (and you'll find the alternate code I had suggested somewhere in there as well) 

Side note : Correction, it looks like I was wrong about changing CPU freq, it can be done (using functions from the ESP-IDF, that is): https://gist.github.com/igrr/a0c3b05f16d12bdd542e0a8d943036e6
But that shouldn't be anything that you're experiencing at all.