Edzelf/ESP32-Radio #101: Get the time from serial command

paullbart, 2018-04-14:
Hi

The time is displayed on the Taft screen. Is it possible to retrieve the time over serial using a command.

Edzelf, 2018-04-14:
Well, it is not a typical radio function.  I could add the time to the test-function.  Maybe you are more interested by a stand-alone sketch that sends the time to the Serial output (or to a display) constantly?

paullbart, 2018-04-15:
what I wanted to do is have a separate Arduino controlled clock that updates its time from the radio. As I know that the esp32 radio can retrieve the time, I can try to build the functionality into your sketch, I just wondered if it was already possible. Thank you for your reply.

Edzelf, 2018-04-15:
If your clock is controlled by an ESP8266 or an ESP32, it would be more logical to get the time trough the network (through a time server).