Edzelf/ESP32-Radio #103: wifi connection drop outs

aber68, 2018-04-16:
hello,

thanks for this great project! i built up the project with node32s end of 2017 and it worked fine. after a recent GIT update of node32 i have now often wifi connection drop outs as well as no stable possibility to program the system via OTA (the device is 3m away from a UPC router). two questions: could this GIT update cause the problem and how can i switch back in GIT?

thanks for advice
andreas / vienna

Edzelf, 2018-04-16:
No idea of how to switch back to an older version.  But you could try to load my binary versin of the sketch.  Then you know if it is software related.

aber68, 2018-04-16:
thanks for your quick answer, i'll give it a try these days ...

aber68, 2018-04-16:
it still doesn't work, get this error ...

Sending invitation to 192.168.0.171 
Uploading.....
22:10:09 [ERROR]: Error Uploading

upload is only possible with phyiscal connection via USB. during playing music and having a drop out the station changes to next, is this a "feature"?

Edzelf, 2018-04-16:
After a number of seconds without connection to the server, the software assumes that the station is dead and  tries to connect to the next preset.
It looks like the WiFi connection is not stable.