Edzelf/ESP32-Radio #104: more lcd / tft / oled support

mikekgr, 2018-04-17:
Dear Sir,
first of all, I would like to express my admiration for your big and fine ESP32 Radio project! Thanks a lot.
The only "problem" that I have, is the TFT LCD that you have selected and I can not find in Ebay. 
Please consider to support OLED based solutions as SSD1306 128X64 ( that many ESP32 has on board ).

Thanks and Best Regards,
Mike Kranidis

locki-cz, 2018-04-18:
Edzelf use TFT ILI9163C it is visible in sketch, also is possible use ST7735 it looks same but it have different chip, 

on aliexpress it is cheaper :) 

how to use ST7735:
https://github.com/Edzelf/ESP32-Radio/pull/51

mikekgr, 2018-04-18:
@locki-cz  T%hanks for your answer, yes I saw it. My suggestion is to cover also -if possible- the OLED and esspecialy SSD1306 controller. That will be very nice. Please consider that now days, many ESP32 has bulid-in OLED screen like this one [ESP32 + OLED](https://www.aliexpress.com/item/ESP32-Bluetooth-WIFI-Kit-0-96-inch-Blue-OLED-Display-Module-CP2102-32M-Flash-IOT-Internet/32825725963.html?spm=a2g0s.9042311.0.0.F9lwOE)  that cost a few dollars ( ~13 )and have everything you want!

Best Regards,
Mike Kranidis

Edzelf, 2018-04-18:
You can order the display [here](https://www.aliexpress.com/item/Mini-1-8-Inch-Serial-SPI-TFT-LCD-Module-Display-with-PCB-Adapter-IC-128x160-Dot/32671985734.html?spm=2114.search0104.3.58.36c543ccTi8Faf&ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10151_10344_10068_5722815_10342_10343_10340_5722915_10341_10698_5722615_10697_10696_10084_10083_10618_10304_10307_10301_5722715_5711215_10059_10534_308_100031_10103_441_10624_10623_10622_5711315_5722515_10621_10620,searchweb201603_19,ppcSwitch_3&algo_expid=e5e5cc52-6b12-4375-bee5-eea30585e241-8&algo_pvid=e5e5cc52-6b12-4375-bee5-eea30585e241&priceBeautifyAB=0).
If you want to use an OLED, you have to make some changes in the software as the screen is much smaller than the one I used.
Note that the radio will work even without the display.

mikekgr, 2018-04-18:
@Edzelf  Dear developer, thanks for your answer. I already ordered one from the link you gave us above although it not say that it is using the ILI9163C controller, I hope to work without code changes.
Sorry to say, I am not coder so I can not modify your code in order to support OLED and for this reason I asked here.
Thanks for all.

Best Regards,
Mike Kranidis

locki-cz, 2018-04-18:
@Edzelf how to edit display info? Can you please put there some hints? I am able show on display info on x,y position with some size about temperature etc ...  in my simple project but your code is different level :) Yes there is D I S P L A Y I N F O section but it looks to complex :( 

It is possible somehow show standby and different screen when it play some radio? I think something like this:
[image]
Maybe also wifi signal level?

Edzelf, 2018-04-18:
Changing info on display is actually very simple.  The screen is divided in three sections, see "scrseg_struct".  The sections are numbered 0..2.  There is an extra fourth section that overlays the third section while handling rotary switch settings.
To send a string to a section, use:
tftset ( uint16_t inx, const char *str ) ;
For example:
tftset ( 1, "Hello there" ) ;
You may also look at lines containing "tftset" and make your changes there.

Edzelf, 2018-04-25:
OLED (128x64) is now supported.

mikekgr, 2018-04-25:
@Edzelf  Thank you very much Sir. 
I will try it one of these days. I think I have to get the latest from your git.

Thanks

bmillier, 2018-05-17:
@Edzelf . Great project and the code is excellent and well- commented.  I only had an ILI9341 LCD on hand (2.8 inch 320x 240). While I couldn't initially get it to work using Adafruit ILI9341 library  when I subbed it into your code, I was able to hook this LCD  up to the HSPI  SPI port, ( you use VSPI port for everything), instantiate an SPI2 port, and then instantiate the adafruit9341 library as   tft2 using that second SPI port.Then everything worked fine.
Thanks for your effort.

ovi4, 2018-07-01:
Fantastic. Thank you very much for the (128x64) OLED support sir. I really appreciate your effort. I wish I could code like you :)

airasz, 2018-07-04:
I used this project as my own radio but At this time i just have 2 available display as choice 1602 and oled 128x32. With 1602 the info just appear on second line contain just station name. Or Can you please add ability to create options on oled library display with 128x32 screen. I will appreciate that. Thank a lot.

Edzelf, 2018-07-04:
Why not buy an 128x64 oled for 2 dollars?

ovi4, 2018-07-04:
I already have  two those screens sir. I will definitely give them a try that's for sure. 
They have been forgotten in one of my drawers and now that you've said it I instantly remembered thank you sir. :)

greppp, 2020-03-01:
Hi @Edzelf, I was wondering if this display is compatible?

https://www.aliexpress.com/item/32798439084.html?spm=a2g0s.12269583.0.0.5bf644f8Y2os2g

(OLED 128x32)

It's about size, to hack a tiny portable radio, i can't use a bigger one.

Thanks for this wonderful project, it is really appreciated to be able to play custom streams on a standalone device!

Edzelf, 2020-03-03:
It need some changes in the software, but I think it will work.
Note the the display space is very limited on this device.