Edzelf/ESP32-Radio #106: Web interface (small) display glitch

kas2, 2018-04-18:
Hi Ed,
Thanks for this great project
I am playing with an ESP32 while waiting for the VS1053 module
For your interest, I noticed this visual glitch
[url=https://postimg.cc/image/vd2vqv2nr/][img]https://s31.postimg.cc/vd2vqv2nr/ESP32_Radio_HTML.png[/img][/url]
No big deal
Thanks again

pcworldhelp, 2018-04-18:
403 ?

kas2, 2018-04-18:
No

kas2, 2018-04-21:
That did the trick
Thanks Ed

polygontwist, 2018-06-17:
Hello,
the better way is to make the button variable, like this (radio_css.h):
``` 
.button {
   min-width: 80px;
  ...
   white-space: nowrap;
}
```
with "min-" the minimum size is defined
Text breaks can be prevented with: white-space: nowrap; 
The buttons are not all the same size, but they are suitable for all languages.

Edzelf, 2018-06-18:
Good suggestion.  I will implement it in the next release.