Edzelf/ESP32-Radio #108: 1.44 ST7735 display

ART032, 2018-04-20:
Hi.
I can't undestend, how change code for use 1.44 ST7735 
i see white screen color
WEB inerface works

PS: test adafrute_st7735 code - work

locki-cz, 2018-04-20:
just edit lines like there: https://github.com/Edzelf/ESP32-Radio/pull/51

ART032, 2018-04-20:
I've already replaced the strings following this link, but it still doesn't work.
my modified code in the attachment.

[Esp32_radio.zip](https://github.com/Edzelf/ESP32-Radio/files/1931253/Esp32_radio.zip)

locki-cz, 2018-04-20:
Here is my edit but it isnt last version, ididnt have time to play with my code for radio for months :( 

[ESP32-Radio_ST7735version.zip](https://github.com/Edzelf/ESP32-Radio/files/1931267/ESP32-Radio_ST7735version.zip)

ART032, 2018-04-20:
it das not work whis that code. 
[image]
[image]

locki-cz, 2018-04-20:
do you have this in defaultprefs.h? you can set ti via web interface, also 1,44" have smaller resolution, than 1,8"

pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"

ART032, 2018-04-20:
yeas, thise is in defaultprefs.h.   in the test adafruit sketh it works, but cut a part of text

a part of defaultprefs.h:

# Clock offset and daylight saving time
clk_server = pool.ntp.org                            # Time server to be used
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_dst = 1                                          # Offset during daylight saving time (hours)
# Some IR codes
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
# GPIO pinnings
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
#
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
#
sd_cs = 21                                           # GPIO Pin number for SD card "CS"
#
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"

ART032, 2018-04-20:
debug: 
no words about start tft  ! why? 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 05 Mar 2018 10:52:00 GMT.  Free memory 171056
D: Partition nvs found, 20480 bytes
D: Read 6 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - Prosyst                   Signal: -67 dBm, Encryption ????, 
D:  2 - PSGuest                   Signal: -67 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: preset with parameter 0
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)

ART032, 2018-04-20:
Why prefs is -1 ????

Edzelf, 2018-04-20:
The setting are missing in the preferences.
You may add them in the web interface like this:
pin_tft_cs = 15      # GPIO TFT CS
pin_tft_dc = 2       # GPIO TFT DC
pin_vs_cs = 5        # GPIO VS1053 CS
pin_vs_dcs = 16      # GPIO VS1053 DCS
pin_vs_dreq = 4      # GPIO VS1053 DREQ

ART032, 2018-04-23:
I directly prescribed in the code PIN. everything worked.

struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",       &ini_block.ir_pin,          35          },
    { "pin_enc_clk",  &ini_block.enc_clk_pin,     25          },
    { "pin_enc_dt",   &ini_block.enc_dt_pin,      26          },
    { "pin_enc_sw",   &ini_block.enc_sw_pin,      27          },
    { "pin_tft_cs",   &ini_block.tft_cs_pin,      15          },
    { "pin_tft_dc",   &ini_block.tft_dc_pin,      2          },
    { "pin_sd_cs",    &ini_block.sd_cs_pin,       21          },
    { "pin_vs_cs",    &ini_block.vs_cs_pin,       VS1053_CS   },
    { "pin_vs_dcs",   &ini_block.vs_dcs_pin,      VS1053_DCS  },
    { "pin_vs_dreq",  &ini_block.vs_dreq_pin,     VS1053_DREQ },
    { "pin_shutdown", &ini_block.vs_shutdown_pin, -1          },
    { "pin_spi_sck",  &ini_block.spi_sck_pin,     18          },
    { "pin_spi_miso", &ini_block.spi_miso_pin,    19          },
    { "pin_spi_mosi", &ini_block.spi_mosi_pin,    23          },
    { NULL,           NULL,                       0           }    // End of list

and I chose the right type of display:

tft->initR(INITR_144GREENTAB);

closed 2018-05-10 by ART032