Edzelf/ESP32-Radio #110: WM8978 instead of VS1053

capedra, 2018-04-24:
Hello guys,

I wonder if it's possible to use codec WM8978 instead of VS1053 to play some music from the Internet. I asked this question to @SensorsIot on YouTube but unfortunately I think that he couldn't read it.

I'm curious about it because if it's really possible then we have a complete board ready for streaming, which is called TTGO TAudio V1.0. The display would be optional since we can adjust all the settings over some Android/iOS App or just a Web App through WiFi or even Bluetooth.
Thanks a lot for your efforts!

Edzelf, 2018-04-24:
Ss far as I understand from the datasheet, the WM8978 is not capable of decoding MP3.  It has an I2S interface, which is no I2C, and is therfore only capable of  decoding I2S streams.

NonaSuomy, 2021-02-17:
https://github.com/CelliesProjects/wm8978-esp32

Edzelf, 2021-02-17:
This projects uses software MP3 conversion.

NonaSuomy, 2021-02-22:
Yup and it works great decodes mp3/AAC/wav/tts.