Edzelf/ESP32-Radio #111: to change the settings

avulyanichev, 2018-04-25:
The default station is 00, the volume is 72. After switching off and on, the volume of the other station and station 04 or 06 may change. We have to fix it again. Help please.

avulyanichev, 2018-04-25:
Specify. When turned off, only the default station and volume settings are changed.

avulyanichev, 2018-04-25:
Excuse me. Another question on the LCD. Many people want to work for 1602 LCD display. Could you add this popular LCD with minimal information output (station number, station name). Thanks.

Edzelf, 2018-04-25:
Volume and station are saved every 10 minutes in order to reduce the number of flash writes.

avulyanichev, 2018-04-25:
the current station and volume remain the same? am I getting this right?

avulyanichev, 2018-04-25:
I understood you. I will check

avulyanichev, 2018-04-28:
remembers once the station and the volume level. you have to change manually.

avulyanichev, 2018-04-28:
Hi. How to disable station memorization in the program?

avulyanichev, 2018-04-28:
clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_21 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
ir_18E7 = downvolume = 3
ir_20DF = station = radio.mvd.ru:8000/mv128.mp3                            #  4 - Miliceiskaya Volna
ir_30CF = upvolume = 3
ir_40BF = station = ic3.101.ru:8000/v11_1?                                 #  2 - Camedy Radio
ir_609F = station = radio40.ru/public/stream.php                           #  6 - Radio_40 Kaluga
ir_6897 = stop
ir_807F = station = dorognoe.hostingradio.ru:8000/radio                    #  1 - Dorognoe Radio
ir_A05F = station = listen23.vdfm.ru:8000/rusradio                         #  5 - Russkoe Radio
ir_A857 = downpreset = 1
ir_C03F = station = ic4.101.ru:8000/v3_1                                   #  3 - Avto Radio
ir_C837 = mute
ir_D02F = uppreset = 1
ir_E01F = station = ep128.hostingradio.ru:8030/ep128                       #  7 - Europa Plus
#
mqqprefix = none
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 4
preset_00 = dorognoe.hostingradio.ru:8000/radio	                   #  0 - Dorognoe Radio
preset_01 = ic3.101.ru:8000/v11_1?                                 #  1 - Camedy Radio
preset_02 = ic4.101.ru:8000/v3_1                                   #  2 - Avto Radio
preset_03 = radio.mvd.ru:8000/mv128.mp3                            #  3 - Miliceiskaya Volna
preset_04 = listen23.vdfm.ru:8000/rusradio                         #  4 - Russkoe Radio
preset_05 = radio40.ru/public/stream.php                           #  5 - Radio_40 Kaluga
preset_06 = ep128.hostingradio.ru:8030/ep128                       #  6 - Europa Plus
#
toneha = 7
tonehf = 1
tonela = 10
tonelf = 10
#
volume = 75
#
wifi_00 = RT-WiFi_3B79/*******

avulyanichev, 2018-04-28:
Saves permanently-preset = 4 and volume = 75.  The sound level saves randomize

Edzelf, 2018-05-08:
1602 display (with I2S backpack) is now supported.

avulyanichev, 2018-05-08:
where can I disable saving settings in my code?

avulyanichev, 2018-05-08:
problem. the universal remote control sends one pulse. The channel and the audio level switch when the second button is pressed. how to change the code?
//**************************************************************************************************
//                                          I S R _ I R                                            *
//**************************************************************************************************
// Interrupts received from VS1838B on every change of the signal.                                 *
// Intervals are 640 or 1640 microseconds for data.  syncpulses are 3400 micros or longer.         *
// Input is complete after 65 level changes.                                                       *
// Only the last 32 level changes are significant and will be handed over to common data.          *
//**************************************************************************************************
void IRAM_ATTR isr_IR()
{
  static uint32_t t0 = 0 ;                           // To get the interval
  uint32_t        t1, intval ;                       // Current time and interval since last change
  static uint32_t ir_locvalue = 0 ;                  // IR code
  static int      ir_loccount ;                      // Length of code
  uint32_t        mask_in = 2 ;                      // Mask input for conversion
  uint16_t        mask_out = 1 ;                     // Mask output for conversion

  t1 = micros() ;                                    // Get current time
  intval = t1 - t0 ;                                 // Compute interval
  t0 = t1 ;                                          // Save for next compare
  if ( ( intval > 300 ) && ( intval < 800 ) )        // Short pulse?
  {
    ir_locvalue = ir_locvalue << 1 ;                 // Shift in a "zero" bit
    ir_loccount++ ;                                  // Count number of received bits
  }
  else if ( ( intval > 1500 ) && ( intval < 1800 ) ) // Long pulse?
  {
    ir_locvalue = ( ir_locvalue << 1 ) + 1 ;         // Shift in a "one" bit
    ir_loccount++ ;                                  // Count number of received bits
  }
  else if ( ir_loccount == 65 )                      // Value is correct after 65 level changes
  {
    while ( mask_in )                                // Convert 32 bits to 16 bits
    {
      if ( ir_locvalue & mask_in )                   // Bit set in pattern?
      {
        ir_value |= mask_out ;                       // Set set bit in result
      }
      mask_in <<= 2 ;                                // Shift input mask 2 positions
      mask_out <<= 1 ;                               // Shift output mask 1 position
    }
    ir_loccount = 0 ;                                // Ready for next input
  }
  else
  {
    ir_locvalue = 0 ;                                // Reset decoding
    ir_loccount = 0 ;
  }
}

Edzelf, 2018-05-08:
Look into the debug output. That will help you to find a solution for your problems.

avulyanichev, 2018-06-06:
I ask you to tell me how to disable saving settings valume and preset. Спасибо.

closed 2018-06-06 by avulyanichev