Edzelf/ESP32-Radio #113: Use of an KMR-1.8 SPI 128*180 TFT

Rebewslin, 2018-05-02:
Hi Edzelf,
what a phantastic project, thanks at first.
Based on your documentation I've bought a KMR-1.8 SPI 128*180 TFT, pic was (nearly) identical to this in your ESP32-radio.pdf.
But it won't work (backlight is on, but only white screen), checked all connections repeated. Configuration is as standard from your .ino:
#define BLUETFT is enabled, others are disabled
Additional Adafruit GFX and ST7735 libraries are installed, sketch compiles without problems.

I've seen (here: [https://forum.arduino.cc/index.php?topic=406501.0]) that this TFT needs other drivers. Is it possible to use it with ESP32-Radio? How?

Thanks!
Nils

Edzelf, 2018-05-02:
I hope your display is 128x160, not 180.
Try to find out what diver works.  Then I can write an include file for this display.

Rebewslin, 2018-05-04:
Hello Edzelf,
Yes, it is 128x160. That was a Typo, sorry. Additional reading of the thread I linked above shows me that there are a lot of different „KMR-1.8 SPI“ displays with several driver ic. I‘ve no possibility to check it out, which one is built inside my display.
I will ask my eBay seller and come back with the answer.

Which display is yours and where exactly can I buy this model?
Nils

fred451, 2018-05-05:
Hello,
I use the same display, mine has an ST7735 chip. I have been able to make it work by using the "TFT_eSPI" library which is available on GitHub.
I had to make some modifications of the original  radio software to make it work correctly. I haven't been able to make the SD working yet.

Rebewslin, 2018-05-05:
Thanks for your informations!
@fred451 : Would you like to share your modifications? How has you included the TFT_eSPI Library?

Rebewslin

fred451, 2018-05-06:
That's ok for me, but the way I did it is not very elegant, well anyway you'll find my mofications in the attached document.

[modradio.docx](https://github.com/Edzelf/ESP32-Radio/files/1977787/modradio.docx)

You might have to make some more modifications as I did remove some parts of the original software which were not necessary for my needs.

Fred

Rebewslin, 2018-05-06:
Hello Fred,

thank you! I‘ll give it a try.

Rebewslin