Edzelf/ESP32-Radio #114: I2S support

vadpol, 2018-05-03:
Hi Ed, would it be possible to include also I2S support into software for external DAC board like pcm5102 for example? This can be used instead of vs1053 however with some sound format limitation (mp3 streams only). From other hand there are still many stations broadcasting in this format and this board is much cheaper and less sophisticated than vs... Many thanks!

Edzelf, 2018-05-03:
I can not write the software for the price of the VS1053.  It is rather complicated to do the MP3 conversion in software.  But if you have the time and skills to make it, I will be very interested.

mikekgr, 2018-05-03:
Dear Sirs, 
I am not coder ( unfortunatelly ) but I this is already done in this git:
https://github.com/earlephilhower/ESP8266Audio
Please check it.
Thanks

vadpol, 2018-05-03:
Well, mp3 conversion requires much less dynamic memory than aac or ogg formats, therefore it is still possible with current hardware setup and external 16 bit (and even internal ESP32 but less 8 bit quality ) DAC. I2S interface is native in ESP32. Unfortunately I am also not that much advanced with such a coding level as I am more to hardware part. But I know that it is possible for sure. Perhaps somebody else could contribute in this feature, that would be great!

pilnikov, 2018-05-03:
karadio32 https://github.com/karawin/Ka-Radio32
mr buddy casino https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

vadpol, 2018-05-03:
Both are closed projects, no flexibility.

pilnikov, 2018-05-03:
non arduino - yes, closed - no

pilnikov, 2018-05-03:
from these projects one can take only output functions in i2s

vadpol, 2018-05-03:
Ok, could you port then the I2S part from these projects?

pilnikov, 2018-05-03:
And why do I need this? I successfully use customized versions of karadio (8266/32) for vs1053 and pcm5102 and I think () that on vs1053 and the sound is better and more possibilities (ogg, aac)

vadpol, 2018-05-03:
Just because edzelf is another project that could have this feature as well. VS1053 is a great chip BUT: 

1. There are too many failing Chinese boards with it on the market and you never know what quality you will get from Ali for example, always a black box
2. In case of portable radio the size of this board (both blue or red) is 2-3 times bigger than pcm
3. Problems with decoupling of the analogue and digital grounds
4. Power consumption of VS is higher (2 built in voltage regulators) which makes battery life shorter
5. Non used features (like recording and built in mic) are making price higher

I like edzelf project because it is open source, very transparent and flexible therefore and really looking for enhancing it with nice features of the choice.

koskee, 2018-05-03:
Note, I had typed out a few paragraphs summarizing all of the reasons that I had previously come across that made this unlikely to happen without some significant changes. 

Then I figured that I should probably at least look at the provided link before I dismissed the idea, but surprisingly I hadn't actually seen the provided info before.

If anyone hasn't clicked the link yet, I recommend it. Without having spent much time with it, it seems like it could be what we have been waiting for? (based on a quick skim of the repositorys feature list )

mikekgr, 2018-05-03:
Dear @koskee ,
which link is that you referenced to?

koskee, 2018-05-03:
@mikekgr sorry, the first one, https://github.com/earlephilhower/ESP8266Audio

mikekgr, 2018-05-03:
ok now it is clear, you mentioned my link. Thanks

vadpol, 2018-05-04:
@Edzelf Ed, might the above library be helpful to develop the relevant code? Many thanks!

pilnikov, 2018-05-04:
I tried this on ESP8266 - hard hangs. Maybe on ESP32 it will be better?

clr666, 2018-06-18:
One man found a way for connecting PCM5102 to VS1053b board, VS has I2S output mode, which need to be initialized by few commands in code. After this both boards plays simultaneously. There is no simple way for connecting one PCM5102, only both boards.