Edzelf/ESP32-Radio #116: listsdtracks() slow

kas2, 2018-05-06:
Hi Ed,

I have an SD card with 877 MP3 tracks
During initialization, Serial Monitor shows "Locate mp3 files on SD, may take a while..."
and display stops for several minutes, waiting for listsdtracks() to return the number of files
then, initialization continues normally
Is there any way to speed up the listsdtracks() job ??
TIA

Edzelf, 2018-05-06:
I don't think so.

kas2, 2018-05-06:
I may have a flaky board
Do you use ESP32_radio with a SD card as a MP3 reader ?
Please confirm the number of tracks on your SD card and the time for listsdtracks() to return the files number
This could be a major no go for MP3 users

Thanks

Edzelf, 2018-05-06:
In my setup (SD card Class 4 with 495 files) it was 32 second.  Good for 20 hours of music.
I think a dedicated MP3 player can do a better job, but hey, it's a radio.

kas2, 2018-05-06:
I now have the feeling that the ESP32 SD library is subpar
With ESP8266 and SdFAT library It does not take more than 1 second for this operation
Let me think about it, I will be back

kas2, 2018-05-07:
So...
I prepared two sketches to reproduce the basic files count operation which is done at line 3110 in       Esp32_radio.ino    (SD_nodecount = listsdtracks ( "/", 0, false ) ;  // Build nodelist)

**SD_espressif_test_1.ino** is based on Espressif SD library
**SdFAT_test_2.ino** is based on Arduino SdFat library (https://github.com/greiman/SdFat)

Using a SanDisk Ultra class10 16GB with 100 MP3 tracks in the root directory
(normally, my SD card is loaded with 3000 files ...)

Results for 100 files, with SDSPEED = 20000000:
 - Espressif SD library: 66228 ms  (1'06") 
 - SdFat library: 771 ms

SdFat library is **85 time faster** than Espressif SD library !!

I attach the two test files for your evaluation (please rename to .ino)
Please let me have your comments
[SD_espressif_test_1.txt](https://github.com/Edzelf/ESP32-Radio/files/1979811/SD_espressif_test_1.txt)
[SdFAT_test_2.txt](https://github.com/Edzelf/ESP32-Radio/files/1979812/SdFAT_test_2.txt)