Edzelf/ESP32-Radio #120: RUS language

ART032, 2018-05-12:
How can I add RUS language. Now it doesn't show any rus symbol. Name of russian song is empty...

Edzelf, 2018-05-12:
Sorry, I don't know.

ART032, 2018-05-14:
i find the linc (http://arduino.ru/forum/programmirovanie/rusifikatsiya-biblioteki-adafruit-gfx-i-vyvod-russkikh-bukv-na-displei-v-kodi)
jast 3 speps: 
1) change glcdfont.c in Adafruit-GFX
2)add in sceth  utf8rus.h: 

 /* Recode russian fonts from UTF-8 to Windows-1251 */

String utf8rus(String source)
{
  int i,k;
  String target;
  unsigned char n;
  char m[2] = { '0', '\0' };

  k = source.length(); i = 0;

  while (i < k) {
    n = source[i]; i++;

    if (n >= 0xC0) {
      switch (n) {
        case 0xD0: {
          n = source[i]; i++;
          if (n == 0x81) { n = 0xA8; break; }
          if (n >= 0x90 && n <= 0xBF) n = n + 0x30;
          break;
        }
        case 0xD1: {
          n = source[i]; i++;
          if (n == 0x91) { n = 0xB8; break; }
          if (n >= 0x80 && n <= 0x8F) n = n + 0x70;
          break;
        }
      }
    }
    m[0] = n; target = target + String(m);
  }
return target;
}

3)Use: display.println(utf8rus("Тест"));  (but i can't find there change code to use thise function

can you show a simple, there is output string to TFT?

Edzelf, 2018-05-14:
Look in the sketch for "utf8ascii" and replace the calls to appropriate calls to functions like "utf8rus".