Edzelf/ESP32-Radio #121: Compile problem

Allhappy, 2018-05-14:
Hello, I´m the only one with problems to compile this new version?

with the version in "archive" folder I can compile in arduino IDE without problems, but with this last version in arduino IDE I have this error... 

"Esp32_radio:5198: error: 'dsp_usesSPI' was not declared in this scope"

Thanks for your project, I am testing it, and it is great!

Thanks,

Edzelf, 2018-05-14:
"dsp_usesSPI" is defined in "Dummytft.h",  "LCD1602.h", "blueftf.h" and "SSD1306.h".
One of those files must be included near line 127 in ESP31_radio.ino.
If you do not use a display, define DUMMYTFT here like this:
```
// Define type of display.  See documentation.
//#define BLUETFT                      // Works also for RED TFT 128x160
//#define OLED                         // 64x128 I2C OLED
#define DUMMYTFT                     // Dummy display
//#define LCD1602I2C                   // LCD 1602 display with I2C backpack
//
```

Allhappy, 2018-05-14:
Thanks for the reply!

I have uncommented the "#define BLUETFT" line, and have the error...

However I have tried uncommented  "#define DUMMYTFT " and the compiling error is the same...
"Esp32_radio:5189: error: 'dsp_usesSPI' was not declared in this scope".

Edzelf, 2018-05-14:
Do you see "Dummytft.h", "LCD1602.h", "blueftf.h" and "SSD1306.h" in your IDE as tabs?
Check if "dsp_usesSPI" is defined in "blueftf.h".
Check if the version defined in ESP32_Radio.ino is like this:
#define VERSION "Fri, 1 May 2018 09:10:00 GMT"

Allhappy, 2018-05-14:
hum.... so...

(Do you see "Dummytft.h", "LCD1602.h", "blueftf.h" and "SSD1306.h" in your IDE as tabs?)
Yes, this tabs are in my arduino IDE.

(Check if "dsp_usesSPI" is defined in "blueftf.h".)
NO... I dont have any "dsp_usesSPI" in "blueftf.h", and I have checked here in Github, and the original file dont have this too...

(Check if the version defined in ESP32_Radio.ino is like this:
#define VERSION "Fri, 1 May 2018 09:10:00 GMT")
This is ok, is the version I have.

Thanks!

Edzelf, 2018-05-14:
Looks like I forgot to upload the latest versions of the *.h files.  I did now.  Could you download them and try again?

Allhappy, 2018-05-14:
ok, for first test I only successfully compile with #DUMMYTFT, but after update the "Adafruit_ST7735" for the latest version,  finely all work in IDE.

I will test it in hardware a few hours later ;)

Best regards!

Rainer-G, 2018-05-18:
Hello, 
what a wonderful project !

I found a similar problem with the latest Adafruit_ST7735.h, ver 1.1, bluetft won't compile
in line 50-52, a third parameter is needed for RST, ver 1.1 does not support 2 parameter for the tft instant 
bool dsp_begin()
{
    tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
                                               ini_block.tft_dc_pin, -1 ) ;            // Create an instant for TFT
 
Best regards, Rainer

mensikv, 2018-05-28:
Hello, I have the same problem as Rainer-G. Calling 
`new Adafruit_ST7735 ( ini_block.tft_cs_pin, ini_block.tft_dc_pin, -1 ) ; // Create an instant for TFT`
fails, because the constructor requiresthree parameters:
```
 Adafruit_ST7735(int8_t cs, int8_t dc, int8_t rst) : 
  Adafruit_ST77xx(cs, dc, rst) {}
```
As RST pin is connected to EN (to reset in reality), I think it needs to be changed to some other pin, that can be supplied as the third parameter.

Have a nice day,
Vlastimil

Edzelf, 2018-05-28:
There are 3 parameters.  The 3rd parameter is "-1", which means that the RST is not used.

mensikv, 2018-05-28:
Well, if I look into bluetft.h, i can see:
```
  tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
                              ini_block.tft_dc_pin ) ;            // Create an instant for TFT
```
So only two parameters and the compilation fails.
I have pulled files yesterday.

Edzelf, 2018-05-28:
You right.  I saw the "-1" in the issue.
I am using version 1.0.8 of the Adafruit ST7735 library.
It has indeed 3 parameters and the last one has a default value of -1.
The headerfile is of ‎20 ‎april ‎2018, ‏‎11:40:55

Rainer-G, 2018-05-28:
Hi,

Version 1.08 has two calls ( 2 or 3 parameter ), Adafruit must have changed it to a single 3 parameter call in Version 1.1. Adding the 3rd parameter with value -1 will make bluetft compile with 1.08 and 1.1

@Vlastimil : there is no need to have a separate output pin for the tft rst input. 
My ESP32 boards ( HUZZAH32 as well as TTGO ) have a dedicated RST pin conneted to the reset button.  On my setup it is conneted to tft RST and vs1053 RST and it works fine.
The EN pin only disables the voltage regulator and is tied to 3V3 on there boards. 

Cheers,

Rainer