Edzelf/ESP32-Radio #124: Display problem - stays white

fetze82, 2018-05-18:
Hello everybody,
Great project, love to build my own one. I made the hardwarestuff today. The Display test "ST7735 grafictest" work fine with the hardware. But i`m not realy know the softwarestuff and i have problems with the display and the ESP32_Radio code.

With the ESP32_Radio, the Diplay stays white.

I use this display: WQScosea Q8S-23 128*160 - ST7735S modul 
[https://de.aliexpress.com/item/WQScosea-Q8S-23-128-160-1-8-1-8-Inch-SPI-Serial-Color-TFT-LCD-Display/32578798873.html?spm=a2g0s.9042311.0.0.5f354c4dRZxRjL](url)

i choose:
tft->initR ( INITR_BLACKTAB ) ;                               // Init TFT interface (160x128)

and i set 
tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin, ini_block.tft_dc_pin, -1 ) ;            // Create an instant for TFT

Do anybody has an idea, why it didn`t work and the display stays white

Thanks a lot
Greatings
Chris

Allhappy, 2018-05-18:
Hello, on first boot is normal display Stay white. You have to access to the web interface and load the default settings and one of this settings is load tft or something similar...

Edzelf, 2018-05-19:
You have to be sure that the pins for CS and DC are in the preferences.  You may do so in the config page of the webinterface.  Restart the radio and look at the debug output with the serial monitor.  Check if the pins match the ones you used for the connection with the display.