Edzelf/ESP32-Radio #125: flash Problem

Feuer001, 2018-05-21:
I have a flash problem.
When I try to flash ESP32 radio I have display problems, e.g. a white display or rainbow pixels. (pin config is ok)
If I flash the radio .bin file then everything is ok.
with which settings was the .bin compiled?
I have the following settings:
ESP32 DEV modules;
default;
DOUT;
80MHz;
4M (32Mb);
115200;
none;
(on Apple Mac)

Edzelf, 2018-05-21:
The settings are okay.  Check if you have the latest version of the libraries and the ESP core.

Feuer001, 2018-05-21:
Yes I have the last versions.
Adafruit gfx 1.2.2
Adafruit ST7735 1.0.8
The display works well, only with the lines I have rainbow pixel. (first the line is green then about 1 second later there are rainbow pixels, but only with the Accu line and the volume line.)

Edzelf, 2018-05-22:
I used Adafruit gfx 1.2.3.

Feuer001, 2018-05-22:
[image]

with Adafruit gfx 1.2.3 the Same problem

Feuer001, 2018-05-22:
If I stop the playback, then the display works fine.
When I start the playback then I have rainbow lines again.

Edzelf, 2018-05-22:
Looks like the CS of the VS1053 interferes with the CS of the display.  Are you sure that they are separated and work correctly?
I saw other users with this problem, but I can not reproduce it.

Feuer001, 2018-05-22:
The hardware must be ok, because the display works with the .bin file

Edzelf, 2018-05-22:
Yes, that's a good argument.  It there a reason not to use the bin file?

Feuer001, 2018-05-22:
yes, I am waiting for a 2.8 "(240 * 320) display ili9341 (for a larger font) or is there any work in progress?

Problem solved!!
I have pin 15 (tft_cs)
changed to pin 22, that's it!

Another question:
Can you still integrate Bluetooth to stream music from mobile phone to esp-radio?

Feuer001, 2018-05-22:
Yes and that on 2x TTGO T1 ESP-32 V1.0.0 Rev1.
But I ordered more ESP32, let's see if they work.

I'm curious if I can do it with the display ...

konektor001, 2018-06-02:
I had the same problem and changing the CS pin to 22 solved me to.
Thanks.