Edzelf/ESP32-Radio #130: IR and Encoder response is slow

konektor001, 2018-05-30:
Everything works fine, but the rotary encoder and IR response is very slow. For example, the double click function is unreliable for this. The simple click is also very slow and unreliable. I tried very short or longer presses but it's the same. For the volume change I have to rotate several rounds for a 10-15 change.
If I check the IR receiver in serial monitor, I have to press a button (with various remote controllers) 3-5 times to catch it. The web interface and MQTT response is OK. Maybe the loop function does a lot of things and that's why it is so slow?
Is this the maximum compute power for the ESP32?
Thanks for your answer and sorry for my english.

Edzelf, 2018-05-30:
Both are handled by interrupt and should be very responsive.

Edzelf, 2018-05-30:
I will check the debug output and show it here later.

Edzelf, 2018-05-31:
This is what you should see in the debug output when you turn the knob very fast:
```
D: Rotation count 10
D: Rotation count 1
D: Rotation count 6
D: Rotation count -1
D: Rotation count -6
D: Rotation count -5
D: Rotation count 3
D: Rotation count 4
D: Rotation count -7
D: Rotation count -4
D: Rotation count -4
D: Rotation count -1
D: Rotation count 7
D: Rotation count -7
```
The volume should change accordingly.  The count is the number of movements detected in the interrupt service routine.
Things like "double click" should also be displayed in the debug output (about 0.5 seconds delay).
It should even work normally while playing a 320 kbps station.

Feuer001, 2018-06-01:
I have the same problem. A normal button also has the problem.
It takes about 1 second until the button is pressed.

Edzelf, 2018-06-01:
What is the function of your "normal button"?

Feuer001, 2018-06-01:
normal button, e.g. start / stop or volume up / down. at the rotary encoder and IR the same, very slow. actually not usable. only mqtt works fine. but if the radio stops for a long time, the play via mqtt will not work. it looks like the interrupt is not working properly.!?

Feuer001, 2018-06-01:
testen on a TTGO T1 and a ESP Devkit v.1

Edzelf, 2018-06-01:
Input pins are scanned max. 10 times per second.  If you are listening to a station with a high bitrate, this may decrease to 5 times per second for 320 kbps.  So you have to press the button for at least 200 msec.
Just tested it with GPIO0 with function "upvolume = 1".  But I cannot reproduce a situation where the scantime is less than once per second.
The scan frequency depends on the time of the main loop ("void loop()").
I added a counter for this in the 01-may version.  It will be visible in the debug output when you issue the "test"-command.  In my situation I see:
`D: scaniocount is 5`
when listening to 320 kbps station.
Could you test this in your setup?

Feuer001, 2018-06-01:
my Output is:
D: Stack maintask is 5896
D: Stack playtask is 1044
D: Stack spftask  is 656
D: ADC reading is 92
D: scaniocount is 6
D: Free memory is 91508, chunks in queue 388, stream 5744, bitrate 172 kbps

Edzelf, 2018-06-01:
Looks normal.  The switch should react in q60 msec.  Please test it while looking at the debug output.

konektor001, 2018-06-02:
With the latest (01-06-2018) version the response is much better. The encoder working fine, only the IR receiver function is a little sluggish, sometimes it receives 2-3 times the command (and after it sometimes crashes the board), other times it receives immediately. Is it possible to implement some compare/delay function not to receive the repeated IR signal eg. for 0.3s?

Feuer001, 2018-06-02:
Buttons react better with the new version. scaniocount is 4-5 at bitrate 192 kbps.
But the ir only works very rarely, if the radio stops working ir without any problems ...

Edzelf, 2018-06-04:
I 've tested the IR.  In my set-up: pin_ir = 35 and using the cheapest I_-control I could fine, I can not reproduce the problems.  I programmed 1 button ("40BF") en pushed all 6 buuons once.  The result in the debug output is: 
```
D: IR code 807F received, but not found in preferences!
D: IR code D02F received, but not found in preferences!
D: IR code 48B7 received, but not found in preferences!
D: IR code 08F7 received, but not found in preferences!
D: IR code E01F received, but not found in preferences!
D: IR code 40BF received. Will execute upvolume = 2
D: Command: upvolume with parameter 2
D: Volume is now 78

```

Edzelf, 2018-06-06:
See my previous reaction.  I cannot reproduce the problem.

avulyanichev, 2018-06-06:
I also have this problem with the remote, only with the programmable.

Feuer001, 2018-06-07:
Ed can you give me your flash config please.
e.g. arduino version, lib version, esp type and config.
It looks like I'm having a problem there ...

Edzelf, 2018-06-07:
I've added the compiler output to the document.  There you can see the library versions.  I used all the latest versions.  I also added the compiled binaries to the "bin"-directory, so you can flash the directly into your ESP32.  I used several ESP types for this radio, today is is an ESP-WROOM-32 (CE 1313).
Current configuration:
```
clk_dst = 1
clk_offset = 1
clk_server = pool.ntp.org
#
gpio_00 = upvolume = 1
gpio_13 = downvolume = 2
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqtttpreset = none
mqttuser = none
#
pin_enc_clk = 25                              # GPIO Pin number for rotary encoder CLK
pin_enc_dt = 26                               # GPIO Pin number for rotary encoder DT
pin_enc_sw = 27                               # GPIO Pin number for rotary encoder SW
pin_ir = 35                                   # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                # GPIO Pin number for SD card CS
pin_tft_cs = 15                               # GPIO Pin number for TFT CS
pin_tft_dc = 2                                # GPIO Pin number for TFT DC
pin_tft_scl = 13                              # GPIO Pin number for SCL
pin_tft_sda = 14                              # GPIO Pin number for SDA
pin_vs_cs = 5                                 # GPIO Pin number for VS1053 CS
pin_vs_dcs = 16                               # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 4                               # GPIO Pin number for VS1053 DREQ
#
preset = 12
preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
preset_12 = 178.79.158.160:8424/stream               # 12 - AT Radio 320 kbps
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 74
#
wifi_00 = NETGEAR-11/*******
wifi_01 = SSID2/*******
```

Edzelf, 2018-06-07:
Very unlikely.  Could you try to load the binary?

fletsche, 2018-06-07:
I think I also have the problem with slow and sometimes missing responses (especially with the rotary encoder). The issue with the IR remote is maybe a bit more complex:
Most remotes I know are sending their code only once per keypress. If you hold the button (i. e. Vol+) it will send a special repeat code. As far as I understand the source these repeat codes are ignored (@Edzelf: correct?). That means that in order for the esp32 to get the code right you have to first point the remote to the sensor and then press the button.

Feuer001, 2018-06-07:
with the binary i have the Same Problem.

My remote only sends once per keypress

Edzelf, 2018-06-07:
Then check your wiring, change the input port, try to make a small program that tests your remote.
@fletsche: the software read any code and compares it to a list of known codes.  Every code will be displayed in the debug output.  In my case, it never misses a code, using all the remotes I have availanle.

Edzelf, 2018-06-09:
Hmm, Interrupt on enc_clk_pin (and not on isr_enc_turn) should work...
But maybe it helps....

dpetric12, 2018-06-10:
I don't use IR, but rotary encoder was very slow. Until I uncomment this line in program. Now it works like a charm! Thank you Ed for a wonderfull project!

Edzelf, 2018-06-10:
I will check if it also works with my encoder.  If so, I will leave the line uncommented.

konektor001, 2018-06-10:
In version 10-may-2018 (where you uncomment "**_//attachInterrupt ( ini_block.enc_dt_pin, isr_enc_turn, CHANGE ) ;_** ") the encoder responses finally as it shoud be, but after you change something with the web interface (preset, volume, stop, ...) or change preset with encoder, it generates -1 values at different tempos, so the volume goes down slowly. If I comment that line like it was before, the error gone, but the response is much slover.

Edzelf, 2018-06-10:
I assume that you mean: the debug output, not the webinterface.  I will test it.  There seem to be different rotary encoders.  I only have one type, a working one....

konektor001, 2018-06-10:
No, I mean in the webbrowser, at http://xxx.xxx.xxx.xxx/config.html. The rotary encoder (the hardware) working fine.

Edzelf, 2018-06-11:
Yes, now I understand.  That means that the rorary switch randomly generates interrupts without touching it!  I think that the WiFi output interferes with the wiring of the rotary encoder.  Once I had such an interference on the reset line of the ESP32.  So please check the wiring of the rotary switch.  Keep it away from the antenna.  Use short wires.  Mayby you can use a shielded  cable.  Or use small capacitors on the 3 signal pins to ground.

konektor001, 2018-06-11:
Maybe it's not an interference, because it occurs only at some specific conditions (what I wrote before: when you send MQTT command "STOP", or after you acces config page on webinterface). When you send other MQTT command (for example preset change) or you acces "control" page on webinterface it's working perfectly.

Edzelf, 2018-06-11:
Maybe...  But it happens when the ESP32 is sending WiFi data.  It can be easily checked if you remove the 3 signal wires of the rotary encoder.  By the way: what are the pins you are using for the encoder.  Maybe you are using a pin that has no pull-up.

konektor001, 2018-06-11:
I use the default 25-26 pins. During the ESP32 streams a radio it receives WIFI data and sends it also (sends MQTT data for example) and there's not **Rotation count -1**.

Edzelf, 2018-06-11:
I rest my case..

konektor001, 2018-06-14:
I've uploaded your BIN file and with that now everything work fine (IR, rotary, webserver, ...), and there isn't any error. I don't understand, becouse I have the latest Arduino IDE and ESP enviroment. Anyway I'm very happy, and thank you for your efforts.

Feuer001, 2018-06-14:
I also flashed the .bin and everything works great.

Ed's Compiler output:
`I also flashed the .bin and everything works great.

Ed's Compiler output:
`esptool.exe --chip esp32 --port
COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --
flash_freq 80m --flash_size detect 0xe000`

My Compiler output:
`esptool" --chip esp32 elf2image --flash_mode "dio" --flash_freq "80m" --flash_size "4MB" -o "/var/folders/sz/z9h9t_c97fd5dnfq8v7v98zm0000gn/T/arduino_build_898426/Esp32_radio.ino.bin" "/var/folders/sz/z9h9t_c97fd5dnfq8v7v98zm0000gn/T/arduino_build_898426/Esp32_radio.ino.elf"
esptool.py v2.3.1`

on my mac.
maybe there are two different ESP enviroment?

Rainer-G, 2018-06-23:
I'm facing the same problem, the reaction to an IR or Enc action is very slow.

So I did some debugging to find where the time gets lost and saw that the call to mp3loop takes between 70 and 1400 msec. 
This seems to depend on the station's bitrate : Radio Gala with 32 kbit/s had a 1400 msec time, AT Radio with 320 kbit/s had max 140 msec. 
The reason behind might be that mp3loop always tries to send as much data to VS1053 as it can get, up to 5744 bytes as far as I can see, (tmpbuff is 6000). 
II the VS1053 only need 32 kbit/s it takes about 1,5 seconds per call ( called twice in main loop ) so there is a significat delay to commands from IR / Enc / Serial etc.
@Edzelf : would it make sense to decrease the tmpbuff size or is the risc too high loosing data even though mp3loop is called more frequently ?

Rainer

Edzelf, 2018-06-23:
1400 msec is way too much.  I will try to reproduce it.  But I doubt, for I do not see issues with IR and rotary switch.
The tempbuff is meant for emptying the stream inputbuffer as fast as possible.  Reading the stream with small buffers slows down the read process.

Rainer-G, 2018-06-23:
try with Radio Gala or a different station with 32 kBit/sec, 
as far as I understand, VS1053 has a 32 Byte input buffer and will require data only as fast as the stream requires. 
With 32 kbit/second, a stream will transfer 4 kBytes per second: The read buffer is 5.744 Bytes so transferring the readbuffer to tmpbuff and sending it with handlebyte_ch will be around 1,4 seconds, or is there anything wrong with my calculation ?
With a smaller tmpbuff, it will transfer less bytes per call resulting in a smaller main loop time and faster reaction to command, if the buffer is not too small, the more frequent calls to mp3loop will still do the job?

Edzelf, 2018-06-23:
Normally, the data is not written to the VS1053, but it is placed in a queue.  But you are free to try a smaller buffer.

Rainer-G, 2018-06-24:
I did some testing last night using stations with bitrates from 32 to 320 kbit/sec
Test results are attached: [loop time.txt](https://github.com/Edzelf/ESP32-Radio/files/2130539/loop.time.txt)

tmpbuff was set to 2048 Bytes ( same size as the FIFO on the VS1053 as I thought this would be the maximum number of byte transferred through the queue at one go ).
Listening to the audio output I could not hear any interruptions but I did not test it with lots of MQTT / Web Interface activities. 
This could be critical if these activities take longer than the VS1053 to empty the FIFO and running out of data.
Response to IR/Enc commands is fast, scaniocount ranges from 7 to 10
Is this a solution or do I miss something I was not aware of?

Edzelf, 2018-06-24:
You have found a good solution.  Later this week I will try the solution to get rid of the 200 ticks wait time in the mp3 loop.  I will let you know the results.

Rainer-G, 2018-06-25:
not being patient enough, I started to play around with the number of xTicksToWait and changed the code to display buffer errors:

        //      xQueueSend ( dataqueue, &outchunk, 200 ) ;       // Send to queue
              if ( xQueueSendToBack ( dataqueue, &outchunk, 200 ) == errQUEUE_FULL ) 
               Serial.println( "dataqueue full");                // Send to queue

Sending from a tmpbuff with 6000 bytes and reducing 200 to 50 gives a lot of errors, reducing tmpbuff to 2048 or even 1024 but let the 200 ticks gives only errors at a station change, 
see file attached [handlebyte xTicksToWait.txt](https://github.com/Edzelf/ESP32-Radio/files/2132497/handlebyte.xTicksToWait.txt)

so it might be a trade-off between tmpbuff size and main loop time, if the buffer is too small, the queue may run out of data if the functions inside the loop take too much time, with 2048 Bytes and 200 ticks I found scaniocount in the range of 4 ( 32 kbit/sec station ) to 10 ( 320 kbit/sec station )
Even more than 200 ticks will do no harm as it will only wait longer if the buffer is full which does not happen unless a station is changed

Edzelf, 2018-06-25:
I just uploaded the new version.  Reading from the input stream is limited to the available space in the data queue.  I also added a test for the maximum duration of the mp3loop() function.  This can be checked by using the "test" function.
The max. duration of mp3loop() is now about 40 msec (was about 800).
Would you be so kind to test the new version?

Rainer-G, 2018-06-25:
I will do so late afternoon and let you know asap

Von: Ed Smallenburg
Gesendet: Montag, 25. Juni 2018 13:34
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] IR and Encoder response is slow (#130)

I just uploaded the new version. Reading from the input stream is limited to the available space in the data queue. I also added a test for the maximum duration of the mp3loop() function. This can be checked by using the "test" function.
The max. duration of mp3loop() is now about 40 msec (was about 800).
Would you be so kind to test the new version?

Rainer-G, 2018-06-25:
excellent job, Ed!
calculation of qspace for maxchunk is a much better way than having fixed values.
Loop time is now between 45 msec on a 32 kbit/sec Station and 25 msec on a 320 kbit/sec station!
Encoder response is excellent!
I'm now looking into a problem with my IR sender/receiver, 
might be a hardware problem or interrupts get lost, 
ir_loccount is set to zero before 65 changes are detected ( intval not in ranges for short/long pulses).

Edzelf, 2018-06-25:
Made the same change for the SD card input.

Edzelf, 2018-07-09:
I do not have the ESP32 folder for the Mac.  You better tell what the problem is, share your preferences and study the debug output.