Edzelf/ESP32-Radio #132: Problems with some m3u stations

fletsche, 2018-06-02:
I have trouble getting some german stations to work. All of the URLs from [NDR](https://www.ndr.de/service/Die-Radio-Livestream-Links,livestreams101.html) or [DLF](http://www.deutschlandradio.de/streaming/dlf.m3u ) obviously don't get parsed correctly. esp32radio just continues with the next preset station. When I use curl to get the contents of the m3u, I get a single line of text which is a stream URL. When I enter that stream URL into esp32radio everything works fine. Seems that these are not standard m3u files?
However these are used by a lot of german radio stations and they seem pretty straight forward to parse. Maybe Edzelf you could include parsing of these?

fletsche, 2018-06-03:
I successfully used curl to find a working stream URL. That's not the problem. I just thought it might be better to support these m3u files directly, since the stations promise to keep those stable while they might change the actual stream URL occasionally.

Edzelf, 2018-06-03:
I will try to fix it.  It looks like the only difference with other m3u files is a missing "end of line".

Edzelf, 2018-06-04:
I 've made the handling of playlistdata NDR-friendly.

fletsche, 2018-06-06:
Works! Thanks a lot.

closed 2018-06-06 by fletsche