Edzelf/ESP32-Radio #137: Hooking up the Adafruit ESP32 ''HUZZAH'' feather board

ovi4, 2018-06-10:
Hi, i was wondering if anyone has ever tried to use an Adafruit ESP32 (code-name ,,HUZZAH32,,) from their ''feather'' line of boards to build this radio with.  It's a small size board that includes a Li-Po battery charger (on-board). and fewer breakout pins than expected !!!  But did not realise at the time of purchase.  See the link to the above board (pinout page) on the official Adafruit website: ( https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts )

Well my problem is that i cannot work-out how on earth to connect the VS1053 module with the ESP32 board since the pins breakout is different than of an ordinary ESP32 Dev board such as the DO IT ESP32 board described on your project. I've been searching the internet (high and low) with no avail.
Here is my problem:
    XRST   pin EN is available so (all good)
    MOSI   since PIN 23 is SDA (marked on the board) but PIN 18 is marked as MOSI instead
                 So, can I use 18 as the board markings suggests or...?
    MISO   pin 19 is available so is (all good)
    SCK     -what PIN to use since PIN 5 is already assigned to SCK (marked on the board)
    XCS     -what PIN to use since PIN 18 is potentially taken by MOSI and PIN 5 is assigned to SCK/CLK 
                 according to the markings on the board. So can I use PIN 5 or...?
    XDCS   -what PIN to use since PIN 16 is also marked as RX (marked on the board)
    DREQ  -what PIN to use since PIN 4 is marked as A5 (and described on Adafruit as PIN Analog 5 IN)

About me: only just started to study about arduino Uno boards+IDE and recently discovered ESP32 boards (So, pretty much an absolute beginner)and so happened that when choosing and ESP32 board did not know which one to buy since the market is flooded with a huge variety of boards and I tooth I'll go with a known brand (such as Adafruit) so I've ordered 3 of them to play with.

From what I understand (by watching Andreas Spies you tube video) the choice of pins needs to be updated on the main radio sketch under (struct iosetting klist) in radio prefs section and also in defaultprefs.h and add my local WiFi credentials (SSID/password) here as well. By the way Andreas is a really nice guy. I find his videos to always be well documented and very instructive indeed. Thumbs up Andreas.

Other questions:
1. I would like to use a rotary encoder but the markings on it is : KEY, S1, S2 instead of CLK, DT,SW. So what do I do?
2. I would like to use a display but at the moment all I have at hand is either : 
   an OLED display 128x128 colour with SPI pins (GND,VCC,SCL,SDA,RES,DC,CS)
   or another OLED display 128x64 colour with SPI pins marked as ( GND, VCC, DO, DI, RES, DC, CS)
    I can only assume that DO is SCL and DI is SDA or am I wrong ? And also how to wire up more than 1 SPI device on one  ESP32 board since the VS1053 board has SPI pins as well. ?
3. if for any technical reason I cannot use any of my OLED (''SPI'') displays then I'm gonna have to use the radio without a display (and just use the html page) until purchase the right type of display but if that's the case what are the required changes in the sketch?

Any help with the above issues is much appreciated. 
PS: since I.m new on GitHub as well I don't even know if I've posted my questions in the right section
If not I would like to apologise in advance for any inconvenience caused.
Thank You

Edzelf, 2018-06-10:
The radio should work normal on an Adafruit Huzzah board.
The idea of the setup is that you do not have to modify any of the software modules.  All pins can be configured in the config page of the web interface.  So the only thing you have to modify are the credentials of your WiFi network in the Esp32_radio_init.ino (tools directory).  Run this sketch once, then run the ESP32_radio.ino sketch.  Now you should be able to connect to te webinterface.

In the webinterface you have to edit all the pins that use are using for the various functions.  Start with the SPI-bus signals and the extra signals for the VS1053. For example:
```
pin_spi_sck = 5
pin_spi_miso = 19
pin_spi_mosi = 18
pin_vs_cs = 12
pin_vs_dcs = 13
pin_vs_dreq = 14

```
Save the settings.

If you have connected the VS1053 acoording the pins defined here, the radio should work.
Then you can configure your display and/or rotary encoder to any of the free pins.

1 - Rotary encoder:  find out what the the signal for the push-switch is.  Use this for pin "pin_enc_sw" . Connect S1 and S2 to the pins that you configure as "pin_enc_dt" and "pin_enc_clk".  If "Volume up" works in the wrong direction: just swap S1 and S2.
2 - The 128x128 display is the best one for the radio.  It will probably not work, but I ordered one and as soon as the package arrives I will make a driver for it.
3 - See 2.

Rainer-G, 2018-06-11:
Hi,

this is my setup for the Adafruit HUZZAH32 and the Adafruit VS1053. 

pin_enc_clk = 36 with external pullup ( see remarks below )
pin_enc_dt = 4
pin_enc_sw = 34 with external pullup
pin_ir = 21
pin_sd_cs = 13
pin_shutdown = 32
pin_spi_miso = 19
pin_spi_mosi = 18
pin_spi_sck = 5
pin_tft_cs = 26
pin_tft_dc = 25
pin_vs_cs = 12
pin_vs_dcs = 15
pin_vs_dreq = 33

I created a PCB to hold all components included the rotary encoder and the IR receiver .
The IO pins were selected to give minimum connection length
I changed the ADC Pin to 35 CHANNEL_7. 
On the HUZZAH32 it is internally connected to the battery and also set the attenuation to 11dB.
The shutdown pin mutes my amp as the VS1053 does some noise whe powering up.

Once I had the board working, I found that GPI36 had an odd behaviour, it permanetly creates interrupts once the adc1_get_raw function is called so I guess I need to change the layout again.

Good luck, Rainer

Rainer-G, 2018-06-11:
some more information re. HUZZAH32:

1. there is a RST Pin which you should use, the EN pin is tied to USB voltage / battery voltage and only enables the voltage regulator, preesind the reset button or downloading code from the IDE will send a reset to the connected devices
2. when powerind is not done via the USB connector but by the USB pin, it must not be more that 5 Volts, the HUZZAH has no Vin pin and the logic around it allowing higher voltages. USB pi is directly connected to the USB connetor so it may feed the PC when voltages are applied to the USB pin. Some computers are not protected from reverse feeding through the USB 

Good luck, Rainer

Rainer-G, 2018-06-12:
@ovi4 

currently. I'm using my Aurex SC-M15/SY-C15 but I'm looking for a D-Amp with a mute input like TPA3110.  The noise from the VS1053 ( more like a "plop" )comes from charging the output capacitors and/or when changing a preset, not from the power supply directly.
The GPI pull up resistors are 39k, the internal ones are aroung 40k so this should do it.
I don't want to publish the PCB but I can send it to you by mail directly

Regards, Rainer