Edzelf/ESP32-Radio #138: Choppy playback at high bitrates

uwe-pt, 2018-06-11:
Playback form stations with high bitrate is choppy. If I go to the web control page and click 'Test' at a high frequency, then playback is just fine. Therefore, I assume the choppy playback has to do something with the buffer. Tried to study the program, but I am completely lost...

Edzelf, 2018-06-11:
Buffering is done by using a queue machanism.  The queue has a length of 400 entries (of 32 bytes).  See line 156 of the source code.
Try "178.79.158.160:8424/stream".  That's a 320 kbps station that should play without distorsions.

uwe-pt, 2018-06-11:
Indeed, the station you suggested plays largely without any problem, but the station 'Magic Oldies Florida' from the default configuration does not. By clicking the 'Test' button at high speed I increase the 'chunks in queue'(Free memory is 68504, chunks in queue **101**, stream 0, bitrate 181 kbps) and then the radio plays smoothly. I tried to increase and decrease the queue length in the source code, but this did not yield any effect. Then again, 'Easy Hits Florida' plays perfectly, even without clicking the 'Test' button.

Edzelf, 2018-06-12:
'Magic Oldies Florida' play correct here.  It is a 192 kbps station.  I cannot understand why the "Test" function has effect on playing the stream.  Under normal circumstances the "chunks in queue" must be near to 400.  If lower, it means that the stream input is slower than the bitrate requires.  After some time the displayed bitrate showed by the "Test" function should be very near to the real bitrate.  A reported bitrate of "181" and "101" chunks in the queue shows that the network cannot handle the stream.
Strange is that the 320 kbps station works correct.  Maybe the network speed varies with the IP of the sender.

uwe-pt, 2018-06-12:
I made a short video to understand the problem better: https://drive.google.com/open?id=1NY_IoD_nClqNlX2yzijODNvioQCc2pzr . You probably can not hear, but you see in the message line when I hit the 'Test' key. Further info: though the video shows clearly the effect of hitting the key, occasionally this has no effect.
My laptop and phone play all stations well.

Edzelf, 2018-06-12:
What I see that sometimes the stream buffer has 0 bytes.  Is it possible for or you to connect to the Internet at another place (neighbour, friend) to see if works there?  It looks like the radio does not like your router.

uwe-pt, 2018-06-12:
Well that could be, I also do not like my router ;-) I'll try next weekend another one. I noticed that the effect mentioned above is even more obvious if I locate the mouse on the 'Test' or 'Status' key and keep than the enter key of my keyboard pressed.

Edzelf, 2018-06-13:
Positioning the mouse cursor at a button and pressing "Enter" is the same as clicking on that button.
The effect is that short TCP packages are generated between the radio and the router.  Maybe in your case that have a positive effect on the packages traveling between the web and the router.   In a normal situation you would only expect a little negative effect on the extra TCP traffic.

uwe-pt, 2018-06-13:
Tested already now using my laptop as hotspot: ESP32-Radio works on all stations perfectly. Time to swap the router... Thank you!

closed 2018-06-13 by uwe-pt

Edzelf, 2018-06-13:
Great!  It would even be greater if you can figure out why.....

uwe-pt, 2018-06-14:
Ok, have tried to do my homework. Did not exactly find out why, but using https://learn.adafruit.com/adabox004/internet-radio the stream is much smoother with only occasional gaps. If gaps occur client.available() returns 0. Behavior stay also the same if the esp32 is just next to the router or in some distance. I would think it hast to do with WiFi.h, but this is just a wild guess.

uwe-pt, 2018-06-16:
Ed, my ISP makes it difficult to use an unbranded router as standalone, therefore I had to find a different solution:
#include <HTTPClient.h>
and in M P 3 L O O P after av = mp3client.available():
      if ( av == 0 )
      {
        http.begin("http://url.of.router");
        http.GET();
      }
Not nice at all, but works for now.

Edzelf, 2018-06-16:
So by just doing some idle TCK access it suddenly works!  Strange, but who cares?