Edzelf/ESP32-Radio #139: Stream to virtul dongle "Blue Pill" - STM32F103C8T6 mini Development Board

gyroing, 2018-06-22:
Hi
i wonder would it be possible to stream chunk mp3 to STM32F103C8T6 the chepest stm32 as virtual mp3 dongle ?
ref:
https://hackaday.io/project/19747-esp8266stm32-wifi-usb-mp3-dongle
https://github.com/satoshinm/pill_serial
If applicable, you can directly connect the radio to any kind of mp3 player from car's usb player to any home entertainment without using vs1053

Edzelf, 2018-06-22:
It looks possible.  But you need a device that can read data from USB-sticks.  That could be a STM32F103 **plus** a VS1053....

gyroing, 2018-06-23:
#139
thank you . but if you see https://hackaday.io/project/19747-esp8266stm32-wifi-usb-mp3-dongle. the mp3 chunks  transfers from pc to es8266 via wifi  and after that esp8266 will send mp3 chunk to stm32  dma enabled (2$ board) . and stm32 emulates virtual mp3 file from its serial port .so each mp3 player decode mp3 and it seems there is no need to use mp3 encoder in microcontroler sids,
your prompt action will be appricated .

gyroing, 2018-06-23:
Duplicate of #139

closed 2018-06-23 by gyroing

Edzelf, 2018-06-23:
My point is, you still need an mp3 decoder, so why not the vs1053.  It's a great device.

gyroing, 2018-06-23:
my considration for car mp3  player without aux . each car media player has a usb  for mp3
. for home appliance i totally agree with you . so i bought 6 vs1053 for your nice project .
but i have cheap 4g hotspots dongle in my car that always i have wifi  . also i saw product that emualted virtual mp3 from a2dp bluetooth to usb mp3 for car that dosn't supprot aux port.
also i think with thid modification i could use prefesional car dac (decoder) to play.
i'm not famillier with esp-idf and i like adruino project . i want to find a way to send client result mp3buffer directly to stm32 and make web radio with virtual mp3 . finnally i think cheap usb mp3 player also could play internetradio
thx

Furkatovich, 2023-03-10:
Hello. Can you please take a video of how the device works and send it.
Thank you.