Edzelf/ESP32-Radio #141: Bug with new Adafruit ST7735 library

Budilnik, 2018-06-28:
First of all, thank you very much for your work. 
With the new library ST7735, a new bug with a volume bar and sometimes with other bars (when i change the station for example) appeared. If I understand correctly, the "fillrect" function does not work correctly. With the ILI9163C library this was not observed.

[https://youtu.be/pFltqMmja5E](https://youtu.be/pFltqMmja5E)

Edzelf, 2018-06-28:
It has probably to do with the CS pin you are using for the display.  If you have a scope or a logic analyzer, you may find that the logic of this pin is inverted.  The pin should be HIGH most of the time.  See issue #100.

Budilnik, 2018-06-28:
Thanks Ed for the solution. With GPIO12 for the tft_cs all works great!

closed 2018-06-28 by Budilnik

Edzelf, 2018-06-28:
Amazing!