Edzelf/ESP32-Radio #142: volume behaviour weirdly

airasz, 2018-07-05:
I've got prequently weird volume change behaviour. Without any trigger, volume decreasing by it self of sometimes drop to zero

Edzelf, 2018-07-05:
Look into the debug output.  That will probably tell you what is going on.

crathje, 2018-07-06:
Any chance you do not have a rotary encoder attached?
I would like n option to disable the rotary encoder input.

I am experiencing something similar and I am getting constant "turns" of the not attached encoder (most likely because I do know the type to order).

 `D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1`

Edzelf, 2018-07-06:
Removing the encoder option is simple.  Just remove the lines:
pin_enc_clk=
pin_enc_dt=
pin_enc_sw=
from the prefences in the config page of the webinterface.

crathje, 2018-07-06:
Shame on me, that was way too simple - did not thought about that and did not look into the code. Just attached decoder to the esp32 this afternoon and did not get time to look deeper into it.

Now it does not decrease the volume any more - great! Thanks!

airasz, 2018-07-07:
Yes. With debug output as my reference the esp act like got command from rotary even though rotary not attached. And the problem solved by comment rotary enc listener on loop. And I think this is unwise solution for me. :-D

avulyanichev, 2018-09-29:
должно закоментировано //vTaskDelete ( NULL ) ;                                          // Will never arrive here

t-c-o-d-e, 2019-04-30:
Dear @Edzelf, I'm experiencing same thing, Rotation count -1 every  even with Rotary Encoder connected and functional. What else should I check? 

> 18:51:18.675 -> D: Rotation count -1
18:51:20.684 -> D: Rotation count -1
18:51:22.767 -> D: Rotation count -1
18:51:26.587 -> D: Rotation count -1
18:51:29.404 -> D: Rotation count -1
18:51:38.410 -> D: Rotation count -1
18:51:39.832 -> D: Rotation count -1
18:51:41.118 -> D: Rotation count -1
18:51:42.959 -> D: Rotation count -1
18:51:48.167 -> D: Rotation count -1
18:51:53.363 -> D: Rotation count -1
18:51:53.779 -> D: Rotation count -1
18:52:01.982 -> D: Rotation count -1
18:52:07.383 -> D: Rotation count -1
18:52:07.795 -> D: Rotation count -1

t-c-o-d-e, 2019-04-30:
Used GPIO12 and GPIO14 for volume Up and Dn fixed this issue.