Edzelf/ESP32-Radio #143: setup touch control

airasz, 2018-07-09:
How to activated touch control?
I just try to activated touch control by register pin on preference/config, extend pin with some short cable.
A pin what I registered not showing when it was reserved on debug, but when I  touch the cable, ESP respond nothing. Any else I missed?

Edzelf, 2018-07-09:
Thera are up to 9 touch pins available, see table touchpin[].  Start your test by adding:
touch_04 = uppreset = 1
to the preferences in the configuration page of the webinterface.  After restart, GPIO4 will act as a touch pin.
Watch the debug output.
Depending on the cable-length the signal may be weak.  You may tune line 2678 for your needs.

airasz, 2018-07-24:
i am still work on touch control but no succes so far. i try add debug tlevel but show nothing.
this is my debug 

D: GPIO35 is LOW, probably no PULL-UP
D: touch_04 pin (GPIO04) is reserved for I/O!

but on pinout map touch4 is on GPIO13
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png
so wich one is correct?

Edzelf, 2018-07-25:
There was a bug in the handling of the touch pins.  I have corrected it.