Edzelf/ESP32-Radio #144: Probably me... crash with realloc() pointer is outside heap areas..

tonyadderley, 2018-07-15:
Hi
Thank you for a brilliant project ( I arrived at this via Andrea Speiss project ) and my apologies if I am doing something silly...

ESP32 continually reboots - log segment follows

D: Headerline: icy-name:NPO Radio1
assertion "heap != NULL && "realloc() pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c", line 278, function: heap_caps_realloc
abort() was called at PC 0x400eed7b on core 1

Backtrace: 0x4008df40:0x3ffd6af0 0x4008e11b:0x3ffd6b10 0x400eed7b:0x3ffd6b30 0x40084675:0x3ffd6b60 0x400846fe:0x3ffd6b80 0x40086331:0x3ffd6ba0 0x4000bedd:0x3ffd6bc0 0x400e07c6:0x3ffd6be0 0x400e0814:0x3ffd6c00 0x400e0835:0x3ffd6c20 0x400e08c1:0x3ffd6c40 0x400d25e1:0x3ffd6c60 0x400d62b9:0x3ffd6c80 0x400d8b8d:0x3ffd6d40 0x400da396:0x3ffd6e00 0x4012ffa0:0x3ffd6e40

Rebooting...
ets Jun  8 2016 00:22:57

Log File Attached - i note at top it says "defaultprefs_html has the wrong version number" - all files downloaded yesterday.... not sure if connected.

Any help appreciated
Tony

[TheErrorIGet.txt](https://github.com/Edzelf/ESP32-Radio/files/2195682/TheErrorIGet.txt)

Edzelf, 2018-07-15:
I will look at the version mismatch tomorrow.  Could you please feed the backtrace into the IDE backtrace analyzer and show the results?

Edzelf, 2018-07-15:
The number in line 3031 should be 180609 instead of 190609.

Edzelf, 2018-07-16:
I corrected the version check, but I could not reproduce your problem.

tonyadderley, 2018-07-16:
Hi - many thanks for your assistance.

I am just trying to work out how to do the IDE backtrace analyser so that I can post that response.

Tony

tonyadderley, 2018-07-16:
hi - the trace results are 

Decoding stack results
0x4008df40: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 139
0x4008e11b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 148
0x400eed67: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40084675: heap_caps_realloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 278
0x400846fe: heap_caps_realloc_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c line 155
0x40086331: _realloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 47
0x400e07b2: String::changeBuffer(unsigned int) at C:\Users\tony\Documents\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp line 170
0x400e0800: String::reserve(unsigned int) at C:\Users\tony\Documents\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp line 158
0x400e0821: String::copy(char const*, unsigned int) at C:\Users\tony\Documents\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp line 195
0x400e08ad: String::operator=(String const&) at C:\Users\tony\Documents\Arduino\hardware\espressif\esp32\cores\esp32\WString.cpp line 238
0x400d25d9: tftset(unsigned short, String&) at C:\Users\tony\Documents\Arduino\ESP32-Radio-master\Esp32_radio/Esp32_radio.ino line 1183
0x400d62b1: handlebyte_ch(unsigned char) at C:\Users\tony\Documents\Arduino\ESP32-Radio-master\Esp32_radio/Esp32_radio.ino line 4290
0x400d8b85: mp3loop() at C:\Users\tony\Documents\Arduino\ESP32-Radio-master\Esp32_radio/Esp32_radio.ino line 3981
0x400da3bd: loop() at C:\Users\tony\Documents\Arduino\ESP32-Radio-master\Esp32_radio/Esp32_radio.ino line 4101
0x4012ff8c: loopTask(void*) at C:\Users\tony\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 18

Edzelf, 2018-07-16:
So there is a problem with tftset().  I have made a correction.  Will upload it.

tonyadderley, 2018-07-16:
Brilliant !!  Now working fine - thank you very much , and especially for the speed of your responses and action.  You are a star !

closed 2018-07-18 by tonyadderley