Edzelf/ESP32-Radio #145: Radio File Support

Arielhh, 2018-07-17:
Hi, 
I am trying to enter stations in the following format but it is not playing 
http://radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
This link for example is not working when using the ESP32 but it is working but it is working when using it via the computer browser.
Can you please help?

Thanks,
Ariel

Edzelf, 2018-07-17:
Did you try the winamp trick as described in the documentation?

Arielhh, 2018-07-17:
I have a Mac and the version i installed was different then the one in the documentation. I couldn't find the menu described in the documentation

Edzelf, 2018-07-17:
I will do it for you tomorrow.

Arielhh, 2018-07-17:
Many thanks :)

Arielhh, 2018-07-17:
I managed to use the Winapp trick but it is giving me the same web address. See attached screenshot:
[image]

Edzelf, 2018-07-18:
The station spec according to WinAmp is:
ndr-ndr1niedersachsen-hannover.cast.addradio.de/ndr/ndr1niedersachsen/hannover/mp3/128/stream.mp3

Arielhh, 2018-07-18:
Hi, it is not the same station.

Edzelf, 2018-07-18:
This one works for me:
radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM

Arielhh, 2018-07-18:
Yes I am using the correct chip VS1053. The stations are located in this server http://radiocast-rr-d.vidnt.com:8000/status.xsl. 
here is the information i am getting through the serial monitor:
Preset is 5
D: Single click
D: STOP requested
D: New preset/file requested (5/0) from http://radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
D: Connect to new host http://radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
D: Connect to http on port 0, extension //radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
D: Request http://radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM failed!
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    33
D:   2 -     0
D:   3 -  6000
D:   4 -     C
D:   5 -  AC45
D:   6 -    C0
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: STOP requested

Do you need the debug log? I need to figure out how to generate it.

Arielhh, 2018-07-18:
[image]

Edzelf, 2018-07-18:
Please remove the "http://" part of the station.

Arielhh, 2018-07-18:
Sorry, i was experimenting. Originally i used it without the http. Please see the same result without the http:
Preset is 5
D: Single click
D: STOP requested
D: Stopping client
D: New preset/file requested (5/0) from radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
D: Connect to new host radiocast-rr-d.vidnt.com:8000/ipbc_IPBCbetLAM
D: Connect to radiocast-rr-d.vidnt.com on port 8000, extension /ipbc_IPBCbetLAM
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/aac
D: audio/aac seen.
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: ice-bitrate=128;ice-channels=2;ice-samplerate=48000
D: Headerline: icy-br:128
D: Headerline: icy-name:no name
D: Headerline: icy-pub:0
D: Headerline: Server: Icecast 2.3.2
D: Headerline: Cache-Control: no-cache
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Duration mp3loop 1002
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    33
D:   2 -     0
D:   3 -  6000
D:   4 -    12
D:   5 -  AC45
D:   6 -    C0
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';

Arielhh, 2018-07-18:
Is there any correlation to the flash frequency (80Mhz insted of 160Mhz) for supporting AAC?
[image]

Edzelf, 2018-07-18:
Flash frequency has no effect on the program.
I cannot reproduce your problem.  Are other (mp3) stations playing normally?
Is there another AAC-station that works?

Arielhh, 2018-07-18:
I looked at the actual chip using a microscope and it seems that it is VS1003B which according to the datasheet is not supporting AAC. I will order the VS1053 and try again. Many thanks for helping!!!

Arielhh, 2018-07-19:
Got the VS1053 all is working fine now. Thanks again for the great instructions and support.

Arielhh, 2018-07-21:
In order to minimize the amount of buttons used. How can i replace the functionality of the Mute to next station (currently supported by pressing the boot button). Mute is not very functional for me. 

Also, to save battery can the screen go black after few minutes and come back to live when turning the rotary encoder? I know it can be done by changing the voltage in pin 7 (BL) at the 1.8" LCD.

Thanks!

Edzelf, 2018-07-21:
You change the functionality of any button.  Read the docuument for details.  For the backlight some extra software (and one ouput pin) is necessary.

Arielhh, 2018-07-21:
Thanks, i will read about the button. For the backlight i think it will be more complicated for me.

Arielhh, 2018-07-21:
Although the program is written in a very organized way i find it difficult to find out how to change the state of the rotary encoder push button so it will change station with each press (the same function of the boot button).  Any ideas?