Edzelf/ESP32-Radio #147: The display does not work.

pierpe83, 2018-07-23:
Hello.
Should the display work without the VS1053 module connected?

Edzelf, 2018-07-23:
I guess so.

Rainer-G, 2018-07-23:
What voltage did you use for VCC on the display ? 3V3 or 5 Volt ?
Did you check the GPIO assignment in the Web Config page ?
I started without the VS1053 as I did not want to waint until it finally arrived and everything was working ( except audio, of course ) but I had to change some pin assignments in the preferences due to using a HUZZA32

pierpe83, 2018-07-23:
The display was powered from 3.3V and later 5V.
I checked the configuration via the web page.
All pins agree with my ESP32 module.

pierpe83, 2018-07-23:
Logging from the serial port?
D: Starting ESP32-radio running on CPU 1 at 240 MHz. Version Mon, 16 July 2018 11:00:00 GMT. Free memory 233876
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 1 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: Added Livebox-Chimes is list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 4
D: 1 - Livebox-Chimes Signal: -65 dBm, Encryption OPEN, Acceptable
D: 2 - Orange_FunSpot Signal: -67 dBm, Encryption OPEN,
D: 3 - AI-THINKER_194932 Signal: -80 dBm, Encryption OPEN,
D: 4 - DMD_M2_tel.664988756 Signal: -89 dBm, Encryption OPEN,
D: End of list
D: Slow SPI, Testing VS1053 read / write registers ...
D: VS1053 not properly installed!
D: Connect to WiFi
D: Try WiFi Livebox-Chimes
D: Connected to Livebox-Chimes
D: IP = 192.168.1.23
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1 / -1 / -1)
D: STOP requested
D: New preset / file requested (99/0) from
D: No host for this preset

Edzelf, 2018-07-23:
GPIO2 is LOW, why is that?
pin_tft_cs set to -1: That means that there is no Chip Select for the TFT.
D: pin_tft_dc set to -1: That means that there is no DC signal for the TFT.
BlueFT need both.
Please add the 2 signals to the configuration.  In my set-up it is:
pin_tft_cs = 15                               # GPIO Pin number for TFT CS
pin_tft_dc = 2                                # GPIO Pin number for TFT DC

pierpe83, 2018-07-23:
Yes yes yes ... The display works.
But this is incomprehensible to me, why do you have to configure these two pins through the Web page, since they are exactly the same configured in the deafultprefs.h file?
Thank you very much.
Now I'm just waiting for VS1053 .....

Edzelf, 2018-07-23:
The defaultprefs are only used for the config page of the webinterface.  It is just text and it gives you a starting point, with some examples.  All  the pins have to be configured, except for the SPI signals.  See the document for details.

ememem13, 2019-04-01:
Please help.
Whatever settings I use, the screen always remains white.

Edzelf, 2019-04-02:
What kind of display are u using?  And show the settings in the config page of the webinterface.

ememem13, 2019-04-02:
Mini 1.8 Inch Serial SPI TFT LCD Module Display with PCB Adapter ST7735B IC

[Serial monitor.txt](https://github.com/Edzelf/ESP32-Radio/files/3032735/Serial.monitor.txt)
[Webinterface.txt](https://github.com/Edzelf/ESP32-Radio/files/3032736/Webinterface.txt)

Edzelf, 2019-04-02:
Settings are okay.  So it must be a wiring error.

ememem13, 2019-04-02:
I checked there was no cabling error.
I checked with another display (KMR-1.8 SPI)
Result unchanged.