Edzelf/ESP32-Radio #148: Problem with the name of the station.

pierpe83, 2018-07-23:
I have a problem with the names of radio stations from a certain channel.
Namely, if I have such a station URL: preset_02 = gr-relay-4.gaduradio.pl/2 # 2 - Impreza
On the display in the name of the station I have only "2", no name "Impreza".
Why?

Edzelf, 2018-07-23:
The "#" acts as a begin of comment.

pierpe83, 2018-07-23:
I know that.
But in every streaming name if there is a "/" then the name on the display jumps after this sign.
This is how my channel configuration looks like:
preset = 11
preset_00 = gr-relay-4.gaduradio.pl/15 # 0 - Dzem
preset_02 = gr-relay-4.gaduradio.pl/2 # 2 - Impreza
preset_01 = 149.202.75.191:8128 # 1 - Polish Radio London
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 # 3 - Magic Oldies Florida
preset_05 = us1.internet-radio.com:8105 # 5 - Classic Rock Florida - SHE Radio
preset_07 = 205.164.62.15:10032 # 7 - 1.FM - GAIA, 64k
preset_08 = stream3.polskieradio.pl:8900 # 8 - Polish Radio 1
preset_09 = stream3.polskieradio.pl:8902 # 9 - Polish Radio 2
preset_10 = stream3.polskieradio.pl:8904 # 10 - Polish Radio 3
preset_11 = stream.open.fm/156 # Classic Rock

There is a problem with channel name on channels 00, 02, 11. It shows the ending number itself with the name of the streaming instead of the name I entered.

Edzelf, 2018-07-24:
gr-relay-4.gaduradio.pl/15 is an "AACP"-stream.  This will not work on a VS1053.
Same for gr-relay-4.gaduradio.pl/2 and stream.open.fm/156.  You can check the stream type in the debug output:
```
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.

```

Edzelf, 2018-07-24:
That's the time-out function kicking in when no data can be decoded.

pierpe83, 2018-07-24:
But why can not it decode data?
At every station I tried, it's the same. Switch several times and then reboot.

Edzelf, 2018-07-24:
Because you don't have a decoder yet.
I don't know why the wifi fails after reboot.  Normally there is no need to reboot.

pierpe83, 2018-07-24:
Thanks that you have so much patience for me. I will probably have more questions in the future.

airasz, 2018-07-27:
I have same issue with wifi failed connect after reboot. In my experience, esp still on connected device list while esp on reboot process. I don't know why Acces point not immediately eliminate the connected esp or maybe esp still hold connection with AP and it make conflict on list in AP. My solution is turn off and wait until esp disappear on list, then turn on again.

Edzelf, 2018-07-27:
I'm suprised that aacp is playing after all.  The station name is taken from the "icy-name", for "gr-relay-4.gaduradio.pl/2" this is:
```
D: Headerline: icy-description:Grupa Wirtualna Polska
D: Headerline: icy-genre:Misc
D: Headerline: icy-name:2
```
So the "2" is displayed.

pierpe83, 2018-07-27:
Edzelf, is there a chance for the radio to display Polish characters?

Edzelf, 2018-07-27:
That's not the plan.