Edzelf/ESP32-Radio #155: IR remote control hits & misses

kas2, 2018-08-01:
Hi Ed,
Using my Pronto IR remote, I get about 70% success.
Most of the time I have to press buttons twice to send a successful command

I changed the bare VS1838 sensor for a VS1838 module (capacitors + pullup) with no additional luck.
Before going further, please confirm your success rate when using an IR remote

TIA

Edzelf, 2018-08-01:
For my cheap remotes there is a 100 percent success at a distance of 2 meters.  I have a 100 uF capacitor on the powerlines of the VS1838, no pull-up.
I have added some debug info for the timing of the IR pulses.  You should see something like:
```
D: IR code B04F received, but not found in preferences!  Timing 576/1658
D: IR code F807 received, but not found in preferences!  Timing 584/1649
D: IR code F807 received, but not found in preferences!  Timing 576/1661
D: IR code F807 received, but not found in preferences!  Timing 583/1652
D: IR code F807 received, but not found in preferences!  Timing 577/1657
D: IR code F807 received, but not found in preferences!  Timing 574/1660
D: IR code F807 received, but not found in preferences!  Timing 580/1654
```
The first number is the averaged duration of a short pulse, the second number is the averaged duration of a long pulse.
If you see big differences in the timing of your remote, change the test values in isr_IR() in the lines:
`  if ( ( intval > 300 ) && ( intval < 800 ) )        // Short pulse?`
and
`  else if ( ( intval > 1400 ) && ( intval < 1900 ) ) // Long pulse?`

kas2, 2018-08-01:
Thanks Ed, I will have a look

avulyanichev, 2018-09-14:
where is the debug code in the latest version?
D: IR code 7A85 received. Will execute upvolume = 2
D: Command: upvolume with parameter 2
D: Volume is now 77
D: IR code B24D received. Will execute downvolume = 2
D: Command: downvolume with parameter 2
D: Volume is now 75
D: IR code B24D received. Will execute downvolume = 2
D: Command: downvolume with parameter 2
D: Volume is now 73
D: Metadata block 64 bytes
D: Streamtitle found, 61 bytes
D: StreamTitle='ЮРИЙ ЛОЗА - ПОЙ,МОЯ ГИТАРА';

avulyanichev, 2018-09-14:
Эд. write a piece of debugging code and where to insert it. Thanks

avulyanichev, 2018-09-29:
Did as you said. no result. if you press vol+ then press once vol-vol + turns on, remembers the last press and turns it on.

Edzelf, 2018-10-01:
There is no memory for pressed buttons, so the received codes have been sent by the IR-remote.  Try a different IR-remote.