Edzelf/ESP32-Radio #158: Touch Control Intitialzation

mikey60, 2018-08-04:
**First, thank you for all your hard work to create a great program!**

In order to get the touch control initialized correctly on the latest version, I had to change the following line of code 2393 in Esp32_radio.ino back to what it was in an earlier version.

 **FROM:**      sprintf ( mykey, "touch_%02d", i ) ;                    // Form key in preferences
     **TO:**       sprintf ( mykey, "touch_%02d", pinnr ) ;            // Form key in  preferences

Also changing the touch THRESHOLD in line 2718 will not affect the  actual threshold used unless code line 2753 is modified.

**FROM:**       level = ( tlevel >= 30 ) ;                              // True if below threshold
    **TO:**        level = ( tlevel >= THRESHOLD ) ;              // True if below threshold

Edzelf, 2018-08-06:
The first issue was already in the 25 july version.
I will make a correction for the second issue in the next release.

mikey60, 2018-08-12:
I still had to change
``` sprintf ( mykey, "touch_%02d", i ) ; // Form key in preferences```
to 
```sprintf ( mykey, "touch_%02d", pinnr ) ; // Form key in preferences```

 in the 06-aug-2018 version to initialize the touch control properly.