Edzelf/ESP32-Radio #161: Cannot Change Time to EST

Alnoor55, 2018-08-13:
Hi
I have changed, ini_block.clk_offset = -5 and ini_block.clk_dst = 1. but the time stays the same as default

Edzelf, 2018-08-13:
Show the lines in your preferences (webinterface, config page).  I hope you did not edit the source code.

Alnoor55, 2018-08-13:
thanks that worked

theblujuice, 2019-06-11:
Heya. I am in a predicament, staying in Adelaide, Australia - we are UTC +9:30 offset making life challenging as you only have the option to offset by an hour. Also our daylight savings in the opposite being Down Under. Any chance you can assist me with a mod to help with my challenging predicament.

Alnoor55, 2019-06-11:
Hi
you have to use telnet commands to set the off set

theblujuice, 2019-06-18:
now I'm even more lost

Rainer-G, 2019-06-18:
Hi,

maybe I can bring a little bit light into the dark:

as you already mentioned, the NTP delivers only UTC time and the program calculates the offset in hours :

      //   clk_offset = <-11..+14>                // Offset with respect to UTC in hours *)              *
      //   clk_dst    = <1..2>                         // Offset during daylight saving time in hours *)      *
      
The offset is an integer variable abd does not allow fractions of an hour

 	 int8_t         clk_offset ;                         // Offset in hours with respect to UTC
   
      if ( argument.indexOf ( "offset" ) > 0 )          // Offset with respect to UTC spec?
    	{
      	ini_block.clk_offset = value.toInt() ;          // Yes, set offset
    	}

And the time lib is configered this way:

      configTime ( ini_block.clk_offset * 3600,  ←
               	         ini_block.clk_dst * 3600,
               	         ini_block.clk_server.c_str() ) ;          // GMT offset, daylight offset in seconds

A way to bypass the problem ( not to solve it ) change 3600 to 1800 and the offset in the config Web page uses the offset in multiple of 0.5 hours, e.g. 19 for an offset of 9:30.

Better would be to change the common code to allow float to clk_offset.

Good luck
Rainer

Von: theblujuice
Gesendet: Dienstag, 18. Juni 2019 11:04
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: Re: [Edzelf/ESP32-Radio] Cannot Change Time to EST (#161)

now I'm even more lost

theblujuice, 2019-06-20:
Heya Rainer-G
Yeah that's how I have done it before using seconds.
Changing the source code makes it difficult for updates and having to go change it each time.
Would be nice if it is set up in time zones like https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv
Daylight savings would work well too. 
For me: const char* Timezone = "ACST-9:30ACDT,M10.1.0,M4.1.0/3"; // Australia/Adelaide
Nice example from https://github.com/G6EJD/ESP32-Miniature-Weather-Station
Not sure if it would work here.