Edzelf/ESP32-Radio #162: Add support for MAX98357 i2S devices

mattdinsdale, 2018-08-14:
Hi can you add support for MAX98357 i2s devices
[https://datasheets.maximintegrated.com/en/ds/MAX98357A-MAX98357B.pdf](url)

Edzelf, 2018-08-15:
No, it's hard to convert mp3 stream to I2S.

mattdinsdale, 2018-08-15:
have you seen
[https://github.com/MrBuddyCasino/ESP32_MP3_Decoder](url) i had some sound but it was jittery but yours has way better features also is arduino compatable

i've managed to move your VS1053 driver into its own sketch/tab and have got the i2s setup commands to compile i just need to work out how send some data over to it from a web stream

Edzelf, 2018-08-15:
I like the word "just"....

K1WIZ, 2021-12-13:
Would love to see I2S support as well.   I built a simple web radio player using this library (https://github.com/schreibfaul1/ESP32-audioI2S) and it plays mp3, AAC, and AAC+ streams really well.  I have it output to a MAX98357 as well and playback is smooth and pop free.  I have thrown both MP3 and AAC+ streams at it.   Your radio project is far more developed, and if include I2S support would be awesome!  :)

Edzelf, 2021-12-13:
See the V2 version.