Edzelf/ESP32-Radio #163: Display glitch - 1.8" 128X160 Display with ST7735S Controller

fetze82, 2018-08-16:
Hey,

I`m using the 1.8" 128X160 SPI TFT LCD Display Modul+ SD Card Display with the ST7735S Controller

The Radio work, but i get some display glitches for example after change the volume. See the picture.

[image]

Does somebody has an idea how to remove this?

Thanks
Chris

Here is my log
ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

D: Starting ESP32-radio running on CPU 1 at 240 MHz. Version Mon, 06 Aug 2018 12:12:32 GMT. Free memory 171672
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 46 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 35 for IR
D: Start display
D: Create list with acceptable WiFi networks
D: Added Chris to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 15
D: 1 - XXXXX Signal: -78 dBm, Encryption WPA_WPA2_PSK, Acceptable

D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: ir_d02f with parameter downvolume = 10
D: Command: ir_e01f with parameter upvolume = 10
D: Command: lstmod with parameter Mon, 06 Aug 2018 12:12:32 GMT
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 00
D: Command: preset_00 with parameter streams.radiobob.de/bob-live/aac-64/mediaplayer
D: Command: preset_01 with parameter streams.radiobob.de/bob-deutsch/mp3-192/mediaplayer
D: Command: preset_02 with parameter streams.radiobob.de/bob-classicrock/mp3-128/mediaplayer
D: Command: preset_03 with parameter streams.radiobob.de/bob-alternative/mp3-192/mediaplayer
D: Command: preset_04 with parameter streams.radiobob.de/bob-bestofrock/mp3-192/mediaplayer
D: Command: preset_05 with parameter streams.radiobob.de/bob-metal/mp3-192/mediaplayer
D: Command: preset_06 with parameter streams.radiobob.de/bob-punk/mp3-192/mediaplayer
D: Command: preset_07 with parameter streams.radiobob.de/bob-blues/mp3-192/mediaplayer
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: preset_12 with parameter us2.internet-radio.com:8050
D: Command: preset_13 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_14 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_15 with parameter us1.internet-radio.com:8105
D: Command: preset_16 with parameter 205.164.62.15:10032
D: Command: sd_cs with parameter 21
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 78
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to XXXX
D: IP = 192.168.178.34
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 18:21:43
D: STOP requested
D: New preset/file requested (0/0) from streams.radiobob.de/bob-live/aac-64/mediaplayer
D: Connect to new host streams.radiobob.de/bob-live/aac-64/mediaplayer
D: Connect to streams.radiobob.de on port 80, extension /bob-live/aac-64/mediaplayer
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Cache-Control: no-cache, must-revalidate
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: Expires: Wed, 15 Aug 2018 16:21:43 GMT
D: Headerline: Location: http://bob.hoerradar.de/aac-radiobob?sABC=5o745317%230%23spr822816552r64sn3n61ss16p16q465%23zrqvncynlre&amsparams=playerid:medi
D: Headerline: Server: Caddy
D: Headerline: Status: 302 Moved Temporarily
D: Headerline: X-Powered-By: PHP/5.6.30
D: Headerline: Content-Length: 0
D: Headerline: Connection: close
D: Switch to DATA, bitrate is 0, metaint is 0
D: Duration mp3loop 40
D: Connect to new host bob.hoerradar.de/aac-radiobob?sABC=5o745317%230%23spr822816552r64sn3n61ss16p16q465%23zrqvncynlre&amsparams=playerid:mediaplayer;s
D: Connect to bob.hoerradar.de on port 80, extension /aac-radiobob?sABC=5o745317%230%23spr822816552r64sn3n61ss16p16q465%23zrqvncynlre&amsparams=playerid
D: Connected to server
D: Switch to HEADER
D: Headerline: icy-br: 64
D: Headerline: icy-pub: 1
D: Headerline: icy-description: RADIO BOB! BOBs Livestream
D: Headerline: icy-audio-info: channels=2;samplerate=44100;bitrate=64
D: Headerline: icy-url: http://www.radiobob.de/
D: Headerline: Instance-id: c03c6cec793576f150c977a82c5a4f51
D: Headerline: Cache-Control: no-cache
D: Headerline: Server: sec-bob.hoerradar.de 8.0.13
D: Headerline: icy-genre: Rock
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint: 16000
D: Headerline: Pragma: no-cache
D: Headerline: icy-name: RADIO BOB! BOBs Livestream
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.
D: Switch to DATA, bitrate is 64, metaint is 16000
D: Duration mp3loop 55
D: Metadata block 48 bytes
D: Streamtitle found, 37 bytes
D: StreamTitle='xxx';

Edzelf, 2018-08-16:
The problem may be the chip select of the display.  This issue is reported elsewhere.  See the other issues.  If you cannot find it, I will search for it tomorrow.  You may also try to use a different pin for the chip select.

fetze82, 2018-08-16:
i use this setting
https://github.com/Edzelf/ESP32-Radio/pull/133/commits/53a9c9aa3a1777cda67f804e8d548f037e5ffc1f

i can`t find an other issue

Edzelf, 2018-08-16:
There is something wrong with pin 15.  See https://github.com/espressif/arduino-esp32/issues/1568.

Edzelf, 2018-08-18:
Do you have a osciloscoop or logic analyser to check the chip select pin?  It should be HIGH most of the time.  Maybe you ca even test it with a led in series with an 1 k resistor or a volt meter.

ART032, 2018-10-18:
Maybe problem is that the past dot of the screen mast have coordinate 127*159? I have the same problem on 1,44 display.

Edzelf, 2018-10-18:
No, the current code will never access points beyond 127 or 159.  Example: assume pos is 100, then the rectangle will start at pixel 100 and have a width of 60 (160-100).  It will write 60 pixels, reaching from pixel 100 to 159.

fetze82, 2018-10-23:
Hey all,
Pleas let me know, if you have a solution.
i just updated to the last version => i got the same effect.
GreatingsChris

Rainer-G, 2018-10-25:
@fetze82 
Hi Chris,
did you use the latest version of the software with the issue #178 fixed ? I had some noise on the audio output when changing stations, this disappeared after applying the bugfix. 
Now I can see some messages when changing stations: 
---> SPI semaphore not taken within 120 ticks by CPU 1, id hspec
The message comes from _void handle_spec()_ and one of the special tasks is updating the display.
Just an idea...
Rainer

fetze82, 2018-11-04:
Hey Rainer,
yes i`m using the latest version, with the fixed ";" problem.
Greatings
Chris

markome, 2019-12-05:
Hello, there is a glitch in a display. I am using 1.8 in 128x160 display with a st7735 controller.
I have tested both CS pin configurations (default 15 and suggested 0).
In both configurations, the display has a glitch. The glitch is as the image is moved to (x, y) = (-1, -2)

[image]

I have also tried swapping for another display module with the same results.
Any suggestions, how to fix the problem?

Cheers!
Marko

andymelichar, 2020-10-25:
If you are having this problem, it's likely you have the wrong display profile set in bluetft.h - the default is "INITR_BLACKTAB" - I commented out that line and uncommented "INITR_GREEENTAB" and the glitch went away:

`
bool dsp_begin()
{
  tft = new Adafruit_ST7735 ( ini_block.tft_cs_pin,
                              ini_block.tft_dc_pin, -1 ) ;        // Create an instant for TFT
  // Uncomment one of the following initR lines for ST7735R displays
  tft->initR ( INITR_GREENTAB ) ;                               // Init TFT interface   <--- This one fixed the glitch
  //tft->initR ( INITR_REDTAB ) ;                                 // Init TFT interface
  //tft->initR ( INITR_BLACKTAB ) ;                                 // Init TFT interface   <--- This is the default
  //tft->initR ( INITR_144GREENTAB ) ;                            // Init TFT interface
  //tft->initR ( INITR_MINI160x80 ) ;                             // Init TFT interface
  //tft->initR ( INITR_BLACKTAB ) ;                               // Init TFT interface (160x128)
  // Uncomment the next line for ST7735B displays
  //tft_initB() ;
  return ( tft != NULL ) ;
}
`