Edzelf/ESP32-Radio #164: some mp3 files classified as text - no playing

tobiasfaust, 2018-08-17:
Hi,
some of my mp3 files has been classified as text so they are dont play. If i try the same url at my laptop / firefox, it will be play

> D: Entry 10 in playlist found: 192.168.10.20/manage/shares/Music/Deutsch/Carl Carlton - Lights Out In Wonderland/10 - Carl Carlton - Lucky Bet.mp3
> D: Connect to new host 192.168.10.20/manage/shares/Music/Deutsch/Carl Carlton - Lights Out In Wonderland/10 - Carl Carlton - Lucky Bet.mp3
> D: Connect to 192.168.10.20 on port 80, extension /manage/shares/Music/Deutsch/Carl Carlton - Lights Out In Wonderland/10 - Carl Carlton - Lucky Bet.mp3
> D: Connected to server
> D: Switch to HEADER
> D: Headerline: Server: Apache
> D: Headerline: X-Frame-Options: SAMEORIGIN
> D: Headerline: Content-Length: 226
> D: Headerline: Connection: close
> D: Headerline: Content-Type: text/html; charset=iso-8859-1
> D: text/html; charset=iso-8859-1 seen.
> D: Switch to DATA, bitrate is 0, metaint is 0
> D: Publish to topic radio/playlist/pos : 10
> D: Publish to topic radio/playing : 1
> D: STOP requested

i cannot find any mistake in my definitions :(

Edzelf, 2018-08-17:
The server is supposed to return something like:
```
D: Headerline: Server: Apache/2
D: Headerline: Last-Modified: Mon, 23 Jan 2017 14:24:55 GMT
D: Headerline: Accept-Ranges: bytes
D: Headerline: Content-Length: 281941
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/mpeg
```
Try to rename you file (and directory) to something without spaces.

tobiasfaust, 2018-08-17:
yes, you are right, there are spaces.
Its possible to fix this in your sketch with brackets or apostrophes?

Edzelf, 2018-08-18:
I guess the spaces must be replaced by "%20".  You can do this in your playlist or make a modification in the sourcecode of the radio.