Edzelf/ESP32-Radio #166: Playback choppy + compile problem + feature request

bobybobybob, 2018-08-18:
The given stations play smoothly. But my favorite stations either have short dropouts or extreme dropouts. On the local PC, these are ok. The display also flickers.  use #define BLUETFT

Audiophile Baroque
http://8.38.78.173:8045/stream

+++++++++++++++++++++++

Dummy Display Mode can not compile.  It work wenn i remove all in displaybattery. Without Display the music still choppy extreme. 

void displaybattery()
{
 
}

++++++++++++++++++++++++

Wifi Signal Streng would be very usefull information on Display and HTML GUI

Rainer-G, 2018-08-18:
ping times do not look very good:

[image]

and stream is empty : 
[image]

on my PC it works as well, but on ESP32 it does not

bobybobybob, 2018-08-18:
My pings are 20ms faster :-)

[image]

Edzelf, 2018-08-18:
Please show the compile error.

bobybobybob, 2018-08-18:
[image]

Edzelf, 2018-08-19:
To get rid of the compile error: go to File->Prfeferences and set "Compiler warnings" to "Default".  For the dummy tft the line 5254 will not be executed, so the error message does not make sense.  The warning can be ignored for the dummy tft.

bobybobybob, 2018-08-19:
Nevertheless, the compiler breaks off. With many details or little details, it makes no difference. Have look on Screenshot. 

When i remove all in 

void displaybattery()
{
//remove all
}

Compiler compile well and happy.

Edzelf, 2018-08-19:
Sorry, can not reproduce your errors here.

bobybobybob, 2018-08-19:
The other one make no compiler brake. Have a look at that screenshots. 

[image]

Maybee a bug in Arduino IDE 1.8.5 ??

Edzelf, 2018-08-19:
The compiler should not check arrayboundaries in default mode.  You may change the line in error to:
`ypos = tftdata [1%TFTSECS].y - 5 ;`

bobybobybob, 2018-08-19:
Perfect, that solve the Problem with the dummylcd mode. 

[image]